christianhaitian/arkos #1174: Power saving

lauren7ino, 2024-09-10:
Hi

On my RGB30 I use the "Switch Power Button to Off" and now I don't have a way to make it sleep.
Q1:
Can you add a shortcut to call the sleep function, even if the "Switch Power Button to Off" is set?
I mean, it's still helpful to have a sleep function without having to shutdown and restart the whole system, specially when commuting between train and subway.

I use the Dim function screen saver.
Q2: The Black screen saver it's the same as a backlight = Off?

christianhaitian, 2024-09-10:
A1: When that feature is enabled, R3+Power is now sleep.
A2: Correct.

closed 2024-09-10 by christianhaitian

lauren7ino, 2024-09-12:
Question:
Is ES suspend <minutes> the same as the sleep function on ArkOS? (light/deep sleep)

christianhaitian, 2024-09-12:
Yes