christianhaitian/arkos #1176: Kodi Crashes When Trying to Watch a Movie

jgrass3, 2024-09-12:
Movies in the Movies Directory cannot be watched with Kodi 21 included with the 08232024 upgrade for the Anbernic RG503,  Kodi immediately crashes back to Arkos.  Intrestingly short videos in the Video directory can be watched, but nothing in the 30-120 minute range.

christianhaitian, 2024-09-12:
I was able to watch a 1 hour and 12 minute movie via Kodi without issue.  The format of the movie I watch was in .mp4 and h.264 encoding.  What encoding and format is your movie in?

jgrass3, 2024-09-12:
Mine is also a MP4 encoded h.264.  I have attached the info file from VideoRedo from which the movie was mastered, and the MediaInfo for the file.  

[image]

[Despicable Me 4 MediaInfo.txt](https://github.com/user-attachments/files/16985378/Despicable.Me.4.MediaInfo.txt)

christianhaitian, 2024-09-12:
Does the movie play via the regular media player in ArkOS?

jgrass3, 2024-09-12:
Yes it does, just fine.  I am now sure it has to do with the length of the movie.  In order to help diagnose, I made a 1 minute cut of the movie and it played fine.  Since yours played at 1:12 in length and mine crashes with a 1:29 length, it looks like the cutoff for playing is somewhere in between.  I will make lengths in between and see if I can come up with a length that causes the crash.  Keep in mind that Kodi 20.5 in the May software handles the full length movie just fine.  So either a Kodi issue or something else in the update is fouled up.

jgrass3, 2024-09-12:
Never mind that last thought.  Just tried to play a 30 minute TV show (Cobra Kai) and it crashes Kodi.  Again, everything plays just fine with the ArkOS media player.  Sorry I can't be of more help.

christianhaitian, 2024-09-13:
Try removing the spaces and special characters in the filename.  See if that makes a difference.

jgrass3, 2024-09-13:
Didn't make a difference at all

christianhaitian, 2024-09-13:
Do you know how to ssh into the unit?  If so, do you see any errors in `/home/ark/.kodi/temp/kodi.log`?  Perhaps you can provide a copy of the log file here.

jgrass3, 2024-09-13:
Ok I'll get the log in a few...

jgrass3, 2024-09-13:
I don't see anything obvious, but then again, I'm never looked at these before.  Maybe something will stand out to you.
[kodi.log](https://github.com/user-attachments/files/16987465/kodi.log)
[kodi.old.log](https://github.com/user-attachments/files/16987466/kodi.old.log)

I just tried to look at another video before I got the log. It was
/roms2/movies/MadMaxFuryRoad.mp4 at 9:42PM.
Good luck.

christianhaitian, 2024-09-13:
I couldn't see anything in the log, however, I was able to replicate the issue by trying to play a video that is a little over 1GB in size.  Most of my videos are under 1GB and why I didn't have the issue.

jgrass3, 2024-09-13:
Ok, glad you can see it.
I am stopping for the night, but what I can do tomorrow is put the May version, play the file in Kodi and send you the log in which Kodi successfully plays the file.  Then you can see what is different.
I really appreciate your support and the effort you are putting into this.  Thank you.

christianhaitian, 2024-09-13:
I found and resolved the issue.  I'll have you test it when you can and confirm it works for you as well.

christianhaitian, 2024-09-13:
As I stated before, I believe I've found the fix and it would be helpful if you can confirm this solves the issue for you as well.

1.  Take the attached compressed file (arkosbackup.tar.gz) that's designated for your device and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1(TF1 Slot).

2.  Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3.  Once completed, launch Kodi and try your media files again that were crashing Kodi before.

Let me know your results.

[arkosbackup.tar.gz](https://github.com/user-attachments/files/16987825/arkosbackup.tar.gz)

jgrass3, 2024-09-13:
WORKS PERFECTLY!  Thanks for this.
For my education, I had a backup folder on both SD's.  I put your file in both of them since I wasn't sure which one to use.
When you are using SD2 for games, which backup folder is used?
Again thanks for helping me out!

christianhaitian, 2024-09-13:
Excellent.  Thanks for reporting this issue and confirming the fix.

christianhaitian, 2024-09-28:
This fix has been included with today's update.

closed 2024-09-28 by christianhaitian

jgrass3, 2024-09-28:
Again, thank you again very much for doing this.  I love having Kodi on my device to watch movies, so you fixing this meant a lot.
Thanks for your support with this.

Joe