christianhaitian/arkos #1180: Löve 2D is not working

dov, 2024-09-15:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANce FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)

I'd like to use the Löve 2D game engine on my R36S, but it fails to run.

### How can the issue be reproduced? (If applicable)

```sh
sudo apt-get install love
mkdir -p ~/tmp/hello-love
cd ~/tmp/hello-love
echo -e 'function love.draw()\n    love.graphics.print("Hello, World!", 640, 480)\nend' > main.lua
love .
```

This fails with the error:
```
Unable to create OpenGL window
This program requires a graphics card and video drivers which support OpenGL 2.1 or OpenGL ES 2.

SDL window creation error: Could not create EGL window surface
```

However, compiling and running the following c-program shows that the version of OpenGL is sufficient:
```c
/* sdl_glinfo.c */
#include <SDL2/SDL.h>
#include <SDL2/SDL_opengles2.h>
#include <stdio.h>

int main(int argc, char* argv[]) {
    if (SDL_Init(SDL_INIT_VIDEO) < 0) {
        printf("Unable to initialize SDL: %s\n", SDL_GetError());
        return 1;
    }

    SDL_Window* window = SDL_CreateWindow("OpenGL ES Info",
                                          SDL_WINDOWPOS_UNDEFINED,
                                          SDL_WINDOWPOS_UNDEFINED,
                                          640, 480,
                                          SDL_WINDOW_OPENGL | SDL_WINDOW_SHOWN);
    if (!window) {
        printf("Unable to create window: %s\n", SDL_GetError());
        SDL_Quit();
        return 1;
    }

    SDL_GLContext context = SDL_GL_CreateContext(window);
    if (!context) {
        printf("Unable to create OpenGL context: %s\n", SDL_GetError());
        SDL_DestroyWindow(window);
        SDL_Quit();
        return 1;
    }

    const char* version = (const char*)glGetString(GL_VERSION);
    printf("OpenGL ES version: %s\n", version);

    SDL_GL_DeleteContext(context);
    SDL_DestroyWindow(window);
    SDL_Quit();
    return 0;
}

```
Compile with:
```sh
gcc -o sdl_gl_info sdl_gl_info.c -lSDL2 -lGLESv2
```
The output of the above program is:
```
OpenGL ES version: OpenGL ES 3.2 v1.r13p0-01rel0.8747d14aec16ec21ac2b2da252c4ec5f
```
### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)

R36S

### What version of ArkOS are you on?

07312024

### Anything else you'd like to include that may help to review this issue or feature request?

I'm currently trying to recompile love to see if I can get it to run.

christianhaitian, 2024-09-15:
While I don't support the R36s, the community build hosted by AeolusX is what you probably want, Love 2D is already included in ArkOS version 10/29/2022 or newer, however, it's the 11.4 version and it has control issues that need to be figured out.

dov, 2024-09-15:
Thanks. I'll check it out.

Meanwhile I managed to compile the HEAD version of love on the R36S. There were only a small number of patches needed.

My branch is in https://github.com/dov/love/tree/r36s .

However, I could net get the joysticks to work in the sample programs. I'll have a look at it.

christianhaitian, 2024-09-15:
awesome.  Hopefully you figure out something.

dov, 2024-09-16:
The joysticks are actually working. Love is simply doing a straight translation of the joysticks events from SDL. See the joystick event and axis table at my R36S programming description page at:

https://github.com/dov/r36s-programming/#mapping-of-the-r36s-controls

Perhaps it would make sense to configure Love to map e.g. the D-Pad to keyboard Up,Down,Left,Right events, etc?

In any case, Love is curently useless in ArkOS in its current state. Would you accept a pull request with my patches?

christianhaitian, 2024-09-28:
Hi, please provide the patches here and I can add them to the build repository.

BoringKate, 2024-10-03:
Maybe relevant: The recent PortMaster release of Balatro (a love2d game) seems to work flawlessly (and uses a more up to date version of love2d).

Several of my own love2d games close randomly when using the built in support for love2d (which I can't really find a cause for) and I was able to "fix" it by just downloading the Balatro port and giving my own love files to it instead of the Balatro one.

I don't know how possible it is to copy their work, but it might fix a bunch of games.

xanthiacoder, 2024-12-08:
I've just started love2d coding for R36S, and so far, so good!

Here's my first open-sourced test with some notes on readme

https://github.com/xanthiacoder/r36s-input-tester

christianhaitian, 2025-01-05:
I've decided to deprecate support for Love2D since I can't get it to work for most games and I've lost interest in further working on it.

closed 2025-01-05 by christianhaitian