christianhaitian/arkos #1181: build mednafen failed

rmtbruce, 2024-09-16:
hi christianhaitian：
i build mednafen failed on the ubuntu 2.0.4.6,chroot arm64.

256.o hash/crc.o minilzo/minilzo.o drivers/libmdfnsdl.a   sexyal/libsexyal.a  -lasound -lm -ldl -lpthread  -lSDL2 libngp.a libsnes.a ss/libss.a ss/libssfplay.a libdesa68.a libmdfnquicklz.a   libzstd.a  libvorbisidec.a  libmpcdec.a  libtrio.a -lFLAC -lz    -lpthread
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/8/../../../aarch64-linux-gnu/libSDL2.so: undefined reference to `wl_egl_window_destroy'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/8/../../../aarch64-linux-gnu/libSDL2.so: undefined reference to `wl_egl_window_get_attached_size'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/8/../../../aarch64-linux-gnu/libSDL2.so: undefined reference to `wl_egl_window_create'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/8/../../../aarch64-linux-gnu/libSDL2.so: undefined reference to `wl_egl_window_resize'
collect2: error: ld returned 1 exit status
make[2]: *** [mednafen] Error 1
make[2]: Leaving directory `/rk3326_core_builds/mednafen/src'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/rk3326_core_builds/mednafen/src'
make: *** [all-recursive] Error 1

need you help
thanks

christianhaitian, 2024-09-16:
Build sdl2 and install it.  Then try again

christianhaitian, 2024-09-18:
Any update on this?

rmtbruce, 2024-09-19:
thanks,has reslove this issue

closed 2024-09-19 by rmtbruce