christianhaitian/arkos #1186: Netplay issue

6040, 2024-09-21:
I have four R36S consoles. Two with "Original Screen" and two with "Type 4".
My goal is to play the original Gameboy Tetris on two consoles against each other via Netplay.
This works fine with the "Type 4" Consoles. The emulator does not matter but both mGBA and TGB Dual work. But it does not work on the original ones.
I use the same TP-Link Dongle on all the consoles and use the same version of Arkos and rom on all.
It says compatible content found and then the activity light on the wifi dongle blinks rapidly and never stops. It never lets me start the two player option.
On the Type 4 consoles the wifi dongle blinks briefly and then it just works.
Any help is gladly appreciated
6040

christianhaitian, 2024-09-21:
Sorry, I don't support the R36s units so I can't provide any guidance on what the issue may be other than make sure all units are using the same version of the retroarch cores.  Those can be updated via retroarch online updater.  Other than that, maybe check with the aeolusux community github for advice if available there. https://github.com/AeolusUX/ArkOS-R3XS

closed 2024-09-21 by christianhaitian

6040, 2024-09-22:
I understand. Could you maybe point me towards some logfiles I should have a look at?

christianhaitian, 2024-09-22:
Unfortunately not for local netplay.