christianhaitian/arkos #1187: ES linking fail (libgo2)

lauren7ino, 2024-09-21:
Hi

Sorry my basic English.

I've been trying to compile ES as in the comment:
https://github.com/christianhaitian/arkos/issues/1179#issuecomment-2351688875

I managed to compile, but I had to include <string> in many .h files by hand.
Maybe some setting I'm missing, but it worked.
I've copied libgo2 by hand as instructed to /usr/local/include/go2.

Now the linker ...

[ 61%] Built target pugixml-static
[ 61%] Built target nanosvg
[ 61%] Built target es-core
[ 61%] Linking CXX executable ../emulationstation
/usr/bin/ld: cannot find -lgo2: No such file or directory
/usr/bin/ld: cannot find -lgo2: No such file or directory
collect2: error: ld returned 1 exit status
make[2]: *** [es-app/CMakeFiles/emulationstation.dir/build.make:861: emulationstation] Error 1
make[1]: *** [CMakeFiles/Makefile2:252: es-app/CMakeFiles/emulationstation.dir/all] Error 2
make: *** [Makefile:156: all] Error 2

The linker fails finding it libgo2.
I've tried everything I could remember, copying go2 to /usr/include/go2, renaming to libgo2, etc. 
I'm out of ideas.

Any suggestion is welcome.
I'm pretty inexperienced compiling and linking in Linux.
I'm using Ubuntu 24.04.1 LTS
Thanks

christianhaitian, 2024-09-22:
Are you trying to build the master branch or the 593noTTS branch?  Try building the 503noTTS branch

lauren7ino, 2024-09-22:
(as as I read somewhere) - Task failed successfully!

593noTTS branch builds and links successfully after I add \#include \<string\> to 6 include files.
Unfortunately the emulationstation generated (4.1 MB) launches as black screen on the rgb30.
(thank god for BaRT!)

I'm now using a new Ubuntu 22.04, as the previous VM went nuts.
I believe the problem may be that it's an ARM64 Linux (as I have a Mac).

But not all is lost.
I have inexpensive Windows AliExpress laptop, and tomorrow I will try again in an Intel Ubuntu.

lauren7ino, 2024-09-22:
Addendum:
Launching es from ssh reports
./es: /lib/aarch64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by ./es)
./es: /lib/aarch64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by ./es)
./es: /lib/aarch64-linux-gnu/libstdc++.so.6: version `CXXABI_1.3.13' not found (required by ./es)
./es: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by ./es)
./es: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by ./es)
./es: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by ./es)

('es' is the name I gave to my emulationstation)

lauren7ino, 2024-09-22:
What's the best course of action? Installing the missing versions on ArkOS?

lauren7ino, 2024-09-23:
Hi again,
I've installed an AMD64 Linux, and tried everything again, with a different luck now.
Now it reports "cannot execute binary file: Exec format error"
It seems I managed to compile an AMD64 version, because it runs in my Linux VM (!), not in ArkOS.

christianhaitian, 2024-09-23:
Have you tried building this in a aarch64 chroot?  Instructions for creating one are here: https://github.com/christianhaitian/arkos/wiki/Building#to-create-debian-based-chroots-in-a-linux-environment

Also, you can build it in an ArkOS install directly on your device as well.  I recommend using a separate sd card for development and following the instructions here: https://github.com/christianhaitian/arkos/wiki/Building-packages-and-modules-on-your-device

lauren7ino, 2024-09-23:
Thanks for the tips!

I've been playing around with this accidental compilation that runs on the Linux AMD64.
I've copied the RGB30 settings and runs in full screen, and except for a few things that don't work (timezone, etc), it's usable enough.

I actually don't need to deploy ES in ArkOS. I only need it to run well enough to see if I can port the code from EmuELEC https://github.com/EmuELEC/emuelec-emulationstation/blob/EmuELEC/es-app/src/CollectionSystemManager.cpp
for Automatic Game Collections.
"Vertical Arcade" and a few common game Genres.

lauren7ino, 2024-09-25:
Hi
Just to let you I almost have no time to work on this (although I would love to!), and most of the time I have is heavily borrowed from my family.
You may close this issue. I will contact you once I have news.
Thanks for you help.

christianhaitian, 2024-09-26:
Ok.  No problem.  Take care.

closed 2024-09-26 by christianhaitian