christianhaitian/arkos #1192: When Bluetooth controller connected then games will react to neither the BT controller NOR standard controls

MeneerJansen, 2024-09-27:
### Describe the Issue (If applicable)
When  a Bluetooth controller is connected (and configured via: _"Start > Configure input"_) then a game (e.g. Mame2003) will react to **neither** the BT controller nor the controls on the device itself (Powkiddy RGB30). 

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)
Powkiddy RGB30

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

ArkOS_RGB30_v2.0_05242024.img

christianhaitian, 2024-09-28:
What bluetooth controller is this?

MeneerJansen, 2024-09-28:
> What bluetooth controller is this?

Xbox One S controller.

christianhaitian, 2024-09-28:
See here https://github.com/christianhaitian/arkos/issues/665

closed 2024-09-28 by christianhaitian