christianhaitian/arkos #1193: 2nd SD card: existing roms, folders etc. get overwritten.

MeneerJansen, 2024-09-28:
### Describe the Issue (If applicable)
Contrary to what is written here:
https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---rgb30#q-how-do-i-make-use-of-the-second-sd-slot-for-roms-and-games
existing roms, folders etc. DO get overwritten. I had an SD card from RockNix with my games on them: it was completely wiped without warning. 

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)
Powkiddy RGB30

### What version of ArkOS are you on?
ArkOS  2.0 05242024

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2024-09-28:
Hmm, that should not have been an issue if you were on 05242024 before you switched to the 2nd sd card.  It was a known issue before the 05242024 release.  Are you sure the contents are not still on the sd card but just in a roms subfolder within the second sd card as required for RCOKNIX?

closed 2024-09-28 by christianhaitian

MeneerJansen, 2024-09-28:
My apologies. Rocknix put all roms on my 2nd SD card in a subdir named "roms". However, ArmOS does not. So it looked like all my old roms were gone, but in fact they were all there in subdirs in the dir "roms".
 
Anyway since you choose to set **all** my issues to "closed" when they are not as well as the unresolved issues by others about (Xbox) bluetooth controllers I don't feel like you are taking issues serious. Good luck in developing and trouble shooting ArkOS.

Bye.

christianhaitian, 2024-09-28:
I handle issues that I can reproduce.  Issues with Xbox controllers are not a concern of mine.  If compatibility with your Xbox controller is a priority for you, this OS is not for you.  Not much to troubleshoot on on ArkOS as it serves is main purpose fine which is emulation gaming on the device with what the chipset can reasonably support.  You will probably best be served with continuing to use your RPI for set top gaming or an Android device..  Good luck on your quest.