christianhaitian/arkos #1194: Restoring ArkOS back-up leads to unbootable device

MeneerJansen, 2024-09-28:
Title says enough. I'm  gonna stop beta-testing ArkOS. Good luck with ironing out the errors in the system.

christianhaitian, 2024-09-28:
Your unit is probably not compatible with ArkOS as these issues doesn't happen on my 2 units.  Also, you never did do OTA update to the latest version of the OS per instructions.  Good luck

closed 2024-09-28 by christianhaitian

MeneerJansen, 2024-09-28:
> Your unit is probably not compatible with ArkOS as these issues doesn't happen on my 2 units. Also, you never did do OTA update to the latest version of the OS per instructions. Good luck

1. Then it surprises me that the Powkiddy RGB30 (a RK3566 based device) does boot w/ ArkOs. 
2.  I downloaded and installed the image file _yesterday_. That must be the most recent version, isn't it?
3.  Like I said in the other issue: I'm afraid you don't take issues as seriously as one could. Developers depend on users for trouble shooting software, OS-es and systems. And yes: we make mistakes a lot of times. But I don't think it's the way forward to to discard them as "closed" issues. Troubleshooting takes time and effort. There is a problem if a system cannot  restore its own backup. 

Good luck developing, testing and trouble shooting ArkOS.

Bye :)

christianhaitian, 2024-09-28:
Restores fine without issue on my end.  I suspect the issue is related to your device.  Again, the image is not always the latest version available.  Updates are provided on a monthly basis via OTA only.  Images are not created as frequently.  Please don't spend time trying to lecture me on how to provide support for free.  I try my best and if it's not good enough for you, please move on and find something else that better fits your needs.

lauren7ino, 2024-09-28:
Your atitude is not acceptable in this community.
@christianhaitian works the ass out for free to make this OS one of the best (if not the best) OS for these devices, and never asked for anything in return.
Go back to the JelOS please.

For reference:
The image provided for download of ArkOS is quite old!
You are supposed to connect it to WiFi and update the OS.
Then you configure/restore the backup.
Learn to read instructions next time.

Notice English is not my native language (Portuguese).
I have an RG353M and an RGB30.
Both work great.