christianhaitian/arkos #1195: Maybe issue

lauren7ino, 2024-09-30:
Hi
Something strange happened to me twice this morning.
I'm setting up my two devices from scratch (different reasons) and twice I did the same.
1. Install IMG fim in SD card
2. Update absolutely everything, including ThemeMaster, PortMaster, RetroArches, etc.
3. Switch to ROMS2
4. In both cases PortMaster launches (from ROMS2) but exits with errors.
6. I fixed it my restoring PortMaster from the ROMS.gz to ROMS2\Tools\PortMaster (posting from memory here)
Suggestion: Either fix what's happening or create a tool/script to restore PortMaster.
I'm not sure if this problem is common enough to do anything about it though.
Thank you for your amazing support! :)

christianhaitian, 2024-09-30:
In both cases, did portmaster already exist on the 2nd sd card before doing the switch to sd2?

lauren7ino, 2024-09-30:
No.
The problem may be that the PortMaster in the first card is way ahead in version (due to auto-updates), compared to the PortMaster extracted to the roms.tar, which is old.
PortMaster may be deploying files in ArkOS that are not in the PortMaster folder.
Just one theory, I can't research right now.

lauren7ino, 2024-09-30:
Another theory, I don't know what the use ROMS2 card script does, but if it's extracting PM from rooms.tag + copying settings from roms\tools\portmaster to roms2\tools\portmaster, probably the new settings + old pm from the tar are not compatible.

christianhaitian, 2024-09-30:
What's probably happening is that the PortMaster files in the roms.tar file are old, however, they should still work to update to a newer version after switching.  No files are copied from Rom1 to Roms2 so it can't be new files are being overwritten with older portmaster files since they wouldn't have existed on roms2 in the first place.

christianhaitian, 2024-09-30:
Which image is this btw?  It might be easier to just update the image with the newer portmaster files.

lauren7ino, 2024-09-30:
> files in the roms.tar file are old, however, they should still work to update to a newer version after switching. 

They do. That's how I fixed PM. Deleted all PM files on roms2\tools\portmaster and expand PM from the tar.

lauren7ino, 2024-09-30:
I agree. File in \roms.tar (using 351files)

christianhaitian, 2024-09-30:
What unit did you do this on?

lauren7ino, 2024-09-30:
rgb30 na rg353m.
Só you want me to do it again for confirmation?
I don’t mind

lauren7ino, 2024-09-30:
The only trick is to postpone changing to roms2 after all updates and port master updates.

christianhaitian, 2024-09-30:
No.  I'll just update the images for them which should hopefully resolve this issue.

christianhaitian, 2024-10-05:
Images have been updated.

closed 2024-10-05 by christianhaitian

lauren7ino, 2024-10-06:
Confirmed with the RG353M. The process that used to fail now works.
Just one detail not critical - The script to change the OS to use SD2 for ROMS, must be modified to Reboot instead of restarting ES. For ex, ThemeMaster is totally missing until a reboot.

lauren7ino, 2024-10-06:
I've tested the RGB30. All fine too.
Did I understand correctly that this new images are ArkOS OTA up to date?
If yes, why don't you take the opportunity to update to update everything in the new images?
PortMaster, ThemeMaster, both RetroArches, all RetroArches OTA updates, shaders, cheats (etc) and installed cores?
This ritual of updating ArkOS and having to go to the other programs by hand to update them is not ideal, I think.

christianhaitian, 2024-10-07:
All of that is updated as part of the process.

lauren7ino, 2024-10-07:
Another minor detail needs a fix:
EmulationStation menu => Advanced => Parse GamesLists toggle.
It must reboot, restarting ES is not sufficient.