christianhaitian/arkos #1196: Not an issue

lauren7ino, 2024-09-30:
Hi
I know I nay be stretching my luck, but maybe refresh the Themes distributed by ArkOS.
Seems to me they are "ancient" and most have fine tuning problems.

es-theme-saga-box + es-theme-sagamodern + es-theme-simply-basic + es-theme-freeplay report the game count in French (main screen) -=> Probable because ES is not in English on my system.

es-theme-saganx + es-theme-nes-box on a 1:1 screen truncates the name of all the systems system (ex: "Station" or "Sega - ...")
es-theme-switch on a 1:1 screen overlaps the systems on the main screen.

There are many nice systems, es-theme-art-book-next-arkos is one of the possible choices.
I also like the simplicity of es-theme-replica (set to yellow) but that's a personal choice.

To test what I'm talking about, try Emulation Station in Portuguese for ex on the RGB30.

christianhaitian, 2024-09-30:
I may switch some of the other themes but not the default one which is nes-box for all devices except for the RGB30.  For the RGB30. the default is sagabox.  By default, sagabox is set to RGB30 in the theme configuration.  If it's not set to that, then you'll get oddities.  The other theme like es-theme-switch just needs to be set to 1:1 in the theme configuration and it should fix the overlap.  Any other suggested themes, I can take a look at.  I don't plan to include the es-theme-art-book-next one as pretty much every other ES based distro makes use of that theme or similar and I want to differentiate ArkOS somewhat from though it is a ES frontend.

closed 2024-09-30 by christianhaitian

lauren7ino, 2024-10-03:
Suggestion:
https://github.com/Vidnez/MinUArk

lauren7ino, 2024-10-03:
Another detail, why are ArkOS boot logos so different between rb353m and the rgb30?
I understand the resolutions are different, but the logo could be the same.
If you want to "want to differentiate ArkOS somewhat", a consistent boot image helps.
Keep up the good work.

christianhaitian, 2024-10-03:
I prefer to differentiate the boot logos because I find it fun to do so.  One of the benefits of this OS is if someone doesn't like the boot logo, it can be easily changed using the logo generator for rk3566 devices or by changing a jpg file for rk3326 devices.