christianhaitian/arkos #1197: Please add RTL8188FU WiFi module

nolt, 2024-10-03:
Hello I have few WiFi adapters based on RTL8188FU which is missing in ArkOS and mine WiFi dongle doesn't works. Could you add this module into ArkOS.
https://github.com/kelebek333/rtl8188fu

Mine device is: R36s
ArkOS version: latest

christianhaitian, 2024-10-03:
Hi, I don't support the r36 devices.  Try requesting this through the community image github here:https://github.com/AeolusUX/ArkOS-R3XS

closed 2024-10-03 by christianhaitian

nolt, 2024-10-04:
Ok that you do not support R36S device but i'm talking about module for WiFi card that doesn't works on ArkOS. On Arch or any other present Linux distribution works of box cuz kernel have this module, ArkOS not.

AeolusUX, 2024-10-04:
This was the first thing I added for the R3XS image so it's been there since the first version of the community image.

nolt, 2024-10-04:
Strange but I didn't have this module. I re-uploaded it https://github.com/AeolusUX/ArkOS-R3XS/issues/67 and works.

Thank you both. 👍