christianhaitian/arkos #1198: DS4 controller configuration for retroarch

jisbruzzi, 2024-10-04:
### Describe the Issue (If applicable)
Similar to https://github.com/christianhaitian/arkos/issues/1064 for the PS4 controller (DualShock 4). Face buttons are mapped wrong, L2 and R2 are not mapped

### How can the issue be reproduced? (If applicable)
1. Pair DS4 via bluetooth
2. Start a psx game
3. face buttons are mapped wrong, L2 and R2 are not mapped

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?
Please replace `~/.config/retroarch/autoconfig/udev/Sony-PlayStation4-DualShock4v2-Controller.cfg`, (same for `retroarch32`) with the file from this gist https://gist.github.com/jisbruzzi/3b559f377074408895ea60964cb71cda

czapa86, 2024-10-12:
Please fix the root of these kind of issues which is:
ArkOs Menu -> Configure Input doesn't save:
input_l_btn = "x"
input_r_btn = "y"
for ALL controllers.
Thanks.

christianhaitian, 2024-10-26:
> ### Describe the Issue (If applicable)
> Similar to #1064 for the PS4 controller (DualShock 4). Face buttons are mapped wrong, L2 and R2 are not mapped
> 
> ### How can the issue be reproduced? (If applicable)
> 1. Pair DS4 via bluetooth
> 2. Start a psx game
> 3. face buttons are mapped wrong, L2 and R2 are not mapped
> 
> ### What device are you using?
> RGB30, should apply for all devices
> 
> ### What version of ArkOS are you on?
> 08232024
> 
> ### Anything else you'd like to include that may help to review this issue or feature request?
> Please replace `~/.config/retroarch/autoconfig/udev/Sony-PlayStation4-DualShock4v2-Controller.cfg`, (same for `retroarch32`) with the file from this gist https://gist.github.com/jisbruzzi/3b559f377074408895ea60964cb71cda

The DS4 mapping you provided has been pushed out with today's update.  Thanks again.

closed 2024-10-26 by christianhaitian