christianhaitian/arkos #1201: Trouble with ArkOS Kernel with Logo Generator

lauren7ino, 2024-10-08:
Hi
I followed the instructions an all appears well, but the final file I don't know what to do with it.

[image]

Every time I burn this image the SD Card becomes unreadable.
RGB30 can't boot it.
Also Z-Zip can't open it.
What Am IA supposed to do with it?

lauren7ino, 2024-10-08:
File: 
[image_RGB30.zip](https://github.com/user-attachments/files/17294121/image_RGB30.zip)

lauren7ino, 2024-10-08:
Images:
https://ibb.co/9h1KFG6
https://ibb.co/VTqYMXt

lauren7ino, 2024-10-08:
Thanks in advance.
I found the original Font and re-did them myself.

lauren7ino, 2024-10-08:
Congrats for the image converter!
I'm a pixel peeper and now I can create pixel-perfect files for my devices.

closed 2024-10-08 by christianhaitian

lauren7ino, 2024-10-08:
For reference, where's the original docx and font.
[ArkOS-RK3566 docx+font.zip](https://github.com/user-attachments/files/17296976/ArkOS-RK3566.docx%2Bfont.zip)

I'm no image expert nor I have the installed tools to manipulate images, so what I did:
I used MS Word to create the negative space font, displayed the Word page at 200% zoom, screenshotted it, and in a Paint.Net analog zoomed it out to fit and changed the page size to the device resolution.

[image]

christianhaitian, 2024-10-08:
It looks nice.