christianhaitian/arkos #1203: Is the support for D-Link DWA-131 (rtl8192eu) planned in the future builds?

nickalleye, 2024-10-10:
Hello! I am a Powkiddy v10 owner on 2.0 (09292024) and I have an issue with the recommended dongle (TP-Link Archer T2U Nano) - it overheats the type-c adapter very quickly. I tried experimenting with different dongles and adapters, and found out that D-Link DWA-140 didn't heat things up. But this thing is huge. So I went and bought DWA-131, which is smaller and it doesn't have 5ghz support, so it shouldn't be too "heavy". 

But ArkOS doesn't seem to recognize it on Powkiddy v10 :(

I checked it on my linux mint laptop and there it works out of the box. So is it possible to add the support for this model in the future?

christianhaitian, 2024-10-10:
Did the T2U Nano get very hot on your Linux Mint laptop as well?  Also, have you looked at any of these other adapters that are confirmed compatible? https://github.com/retrogamehandheld/oga/wiki/Frequently-Asked-Questions#what-wifi-adapters-work

nickalleye, 2024-10-10:
It's not the dongle that's getting hot, it's the typeA>typeC adapter. I had the exact same issue with my previous device (Anbernic rg351p), and I just replaced the stock adapter with another one: the issue went away. So with PKv10 I bought the very same adapter as before, but the issue remained. I am pretty sure it's a hardware issue.

I guess even if ArkOs supported rtl8192eu, it wouldn't guarantee to not overheat the adapter, buuut I could try an alternative OS on a separate SD card and see if 
a) dwa-131 is supported
b) and if yes, does it overheat the A>C adapter

That alternative OS is not very good from my experience. But I will still do it for science!

nickalleye, 2024-10-10:
Ok, so I tried an alternative OS, and it does support dwa-131, but it heats up the type A > type C adapter all the same, so it wouldn't matter. I guess I need to find a better adapter. 
So if there are no plans of dwa-131 support, I guess the issue may be closed

closed 2024-10-10 by christianhaitian