christianhaitian/arkos #1204: Phantom analog inputs

lauren7ino, 2024-10-11:
Hi Chistian,

My RGB30 right analog stick has phantom inputs (my fault).
If I unplug it, Reproarch cursor won't move (reason unknown).

The same way you gave me a clue to make a system wide A-B switch way Xbox style by editing the DTB (a hack that I never went back!)

Do you know it it's possible to manipulate the DTB file do disable the analog sticks?
(maybe comparing it with a DTB files from a device without 1 or 2 sticks?)
Alternative: Resize the dead zone until the problem subsides?

Thank you in advance

christianhaitian, 2024-10-11:
Either try disabling one of the cases here: https://github.com/christianhaitian/RG353VKernel/blob/7dc1d317bce2f91d810e93873c473e91e7fd1e51/drivers/input/joystick/singleadcjoy.c#L168

or modify the deadzones here: https://github.com/christianhaitian/RG353VKernel/blob/7dc1d317bce2f91d810e93873c473e91e7fd1e51/arch/arm64/boot/dts/rockchip/rk3566-rgb30.dts#L388

Might be easier to modify the deadzones in the dtb.  Good luck

closed 2024-10-11 by christianhaitian

lauren7ino, 2024-10-11:
For future reference:
https://discord.com/channels/741895796315914271/1152383241303625758/1294391002026610689

lauren7ino, 2024-10-11:
Do you know if it’s possible to compile the DTS in ArkOS via SSH?

christianhaitian, 2024-10-12:
yes it is.

On an Ubuntu machine, make sure to install `device-tree-compiler` from the apt repo.  Then you can do:
`dtc -I dtb -O dts <dtb file> > <dts text file name>`
For example `dtc -I dtb -O dts rk3566-OC.dtb > rk3566-OC.dts`
Then edit the file with a text editor.
When done, do
`dtc -I dts -O dtb -f rk3566-OC.dtc -o rk3566-OC.dtb`

lauren7ino, 2024-10-13:
OK I have good news.
On line 368 a deadzone value of \<0xff\> gives me very good results.
No longer suffer from ghost inputs (as far as I can tell at this late hour of the night),
so the console it much more usable.
Unfortunately it's not perfect. Right stick up doesn't register, and fake directions keep popping up on the right stick, but now I have to move it.
So, ghost directions seem to have disappeared.
Ports Doom, Quake, Tomb Raider (look around) and Amiga games that need mouse are not really usable, but it's a problem I can live with i the next few weeks.
Maybe I will purchase a couple cheap hall effect sticks and who knows I get lucky? :)

lauren7ino, 2024-10-13:
On a related subject:

The line 388 you recommended (around it, I know) took me in the wrong path but with a nice consequence:
Almost (all?) these device suffer from analog sticks that are "not very analogy".
They seem to have acceleration, which makes very (really, very) difficult small fine movements of a pointer on a ScummVM game.

BUT these settings "abs_x-p-tuning" & Co. when set to zero remove all acceleration.
Very interesting effect!
Maybe these settings were copied from the original Anbernic firmware and they are bad settings we've been copying all along.
(just a theory)

---
By the way, thank you thank you very much for the Options > Advanced > Controller Tester.
I only wish there was a way to clear the screen between outputs, because it's very hard to read with all that scrolling.

lauren7ino, 2024-10-14:
Addendum:
Ghost inputs returned on line 368 with deadzone value of <0xff>,
but I moved the sticks and rebooted and went way in the next boot.
I can live it. Don you if I can go further than 0xff on this setting?