christianhaitian/arkos #1205: 4 very small details

lauren7ino, 2024-10-13:
On a new installation of ArkOS:
1. You are distributing a ScummVM game called Lancelot, which don't seem work out of the box.
2. You are distributing some kind of Wifi usb-card on the Wifi tool.
3. I've talked to the es-theme-replica theme creator, and he says that his theme may be the only one that currently have logos for all the consoles ArkOS supports. Maybe give it a second look, in case you update some ArkOS themes.
4. I can't find any bugs on the new ThemeMaster preview. Maybe make if official in the next ArkOS so people can see the themes before installing? Nobody activates the "install previews", so nobody benefits from this.
5. Thanks, keep up the good work!

christianhaitian, 2024-10-13:
1.  There's no game included with ScummVM at all.  Not sure how you're getting that.
2.  No wifi usb-card is included in the wifi tool.  Just a default ethernet adapter which is normal.

lauren7ino, 2024-10-13:
I believe you.
Maybe it's a mix of my games and a bad result of the scan for new games script ...

christianhaitian, 2024-10-13:
Ahh, you scraped menu.scummvm.  Looks like the scraper thinks it's the game Lancelot. :)

lauren7ino, 2024-10-14:
Thank for the investigation. I barely have time to play lately.
Can you add that "menu.scummvm" to the exclusions list of the screen scrapper?
Thanks

christianhaitian, 2024-10-15:
Yes.  I'll include this fix with the next update.

lauren7ino, 2024-10-15:
What about the TM preview still showing a black screen in the RG353M?
Can you reproduce it?
(RGB30 is fine)

christianhaitian, 2024-10-15:
Yes it's showing a black screen on my RG353M.  I'll try doing some testing on my end to see if I can figure out why.

christianhaitian, 2024-10-16:
Try replacing the file `/roms/tools/ThemeMaster/image-viewer-legacy` with the one in the attached zip file and verify it works for you on both your rg353m and rgb30 units.  Let me know your results.

[image-viewer-legacy.zip](https://github.com/user-attachments/files/17388802/image-viewer-legacy.zip)

lauren7ino, 2024-10-16:
Confirmed this file shows previews both on rg353m and rgb30 👍🏼

christianhaitian, 2024-10-16:
Thank you for confirming.

clach04, 2024-10-20:
@lauren7ino for number 3, is that in reference to https://github.com/dani7959/es-theme-replica? Thanks for sharing, going to check that out :-)

lauren7ino, 2024-10-20:
Yes it is. It's a simple theme with no art inspired in the ZX Spectrum. Uses the Gar Games credits font. Yellow text is nice.

christianhaitian, 2024-10-26:
The emulationstation issue has been corrected with today's update.

closed 2024-10-26 by christianhaitian