christianhaitian/arkos #1206: ScummVM games keybard

lauren7ino, 2024-10-13:
Hi
The keyboard in ScummVM games is completely weird.
Launch any game, and press Start to open the keyboard.
We press for ex the 'x' key and if inputs '[x]', the Delete key inputs [Delete], etc.
It seems some kind error, it writes the key name between [] instead of the key code.
Any chance this bug is fixed somewhere else?
I'm using the default emulator which I believe is standalone.
Thank you

christianhaitian, 2024-10-13:
That's expected.  It's how the scummvm creators designed it.  In order to send the input, you must press the green checkmark button to send the keys to the game.

closed 2024-10-13 by christianhaitian

lauren7ino, 2024-10-13:
Wow. OK ... I always wondered wha the checkmark was for. Thanks!