christianhaitian/arkos #1207: No gamepads detected

TheColin21, 2024-10-19:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANce FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?
yes

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?
yes
### Describe the Issue (If applicable)
A clear and concise description of what the bug is.
I recently updated my RG351V (with V2 screen and the apropriate [dtb file](https://github.com/christianhaitian/arkos/issues/662#issuecomment-1511860062) if that's of interest) to 09292024 from 08232024.
When I boot up the device it now shows "No gamepads detected".
The only working keys seem to be the volume, power and reset buttons.

I've read in this [diff ](https://github.com/christianhaitian/arkos/commit/f606103d023196d14bb42de1a13e9138e7d955c5) that there were some adjustments that sound like a similar issue.

How do I get my Anbernic working again?

### How can the issue be reproduced? (If applicable)

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)

RG351V

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

09292024 

### Anything else you'd like to include that may help to review this issue or feature request?

TheColin21, 2024-10-19:
Even weirder: I connected a USB-C hub with a USB keyboard to the second USB-C slot and it started up normally. All buttons seem to work.
When I disconnect the hub and restart, I get "no gamepads detected" again.

TheColin21, 2024-10-19:
I am able to access the console with the USB-hub and keyboard.
I _did_ update from 08232024 to 09292024 and I'll now look into the SDL files changed within this update.

TheColin21, 2024-10-19:
**workaround:** I took a look at the Update-RG351V.sh and saw, that the last change to SDL files was made in the 05242024 version.

I now linked the SDL files back to these versions, as the may update did (see "Install and link new SDL 2.0.3000.3 (aka SDL 2.0.30.3)"):
`
```
sudo ln -sfv /usr/lib/aarch64-linux-gnu/libSDL2.so /usr/lib/aarch64-linux-gnu/libSDL2-2.0.so.0
sudo ln -sfv /usr/lib/aarch64-linux-gnu/libSDL2-2.0.so.0.3000.3 /usr/lib/aarch64-linux-gnu/libSDL2.so
sudo ln -sfv /usr/lib/arm-linux-gnueabihf/libSDL2.so /usr/lib/arm-linux-gnueabihf/libSDL2-2.0.so.0
sudo ln -sfv /usr/lib/arm-linux-gnueabihf/libSDL2-2.0.so.0.3000.3 /usr/lib/arm-linux-gnueabihf/libSDL2.so
```

Everything is working fine again.

christianhaitian, 2024-10-19:
I have a v1 unit and the latest update works fine on my unit.  Did you make sure your unit was sufficiently charged?  Low battery can cause the gamepad to not be detected on the 351v units.

TheColin21, 2024-10-19:
It was charged around 80%

christianhaitian, 2024-10-19:
Do you have a spare sd card that you can load the 351v image and update that and see if the issue happens again?

TheColin21, 2024-10-19:
Will do👍🏻

TheColin21, 2024-10-19:
I now did a fresh ArkOS install on a Samsung 256GB Pro microSD from the latest image, replaced the dtb and updated to the latest version.
That all worked flawlessly.

christianhaitian, 2024-10-19:
Ok.  I suspect it was the occasional no gamepad bug that the 351v sometimes has.  The 351p and 351m units also had that issue but it was much worse on the 351p/m units versus the 351v.  Thanks for confirming the updates works fine with the latest update on the v2 unit.

closed 2024-10-19 by christianhaitian

TheColin21, 2024-10-19:
I hope it won't break again on the next update😅

christianhaitian, 2024-10-19:
I doubt it breaks if there's a future sdl2 update.  It will just be an issue that may require a few reboots.  Typical for this unit since it has a usb controller.

TheColin21, 2024-10-20:
I can change the links again to test it but I am pretty sure it was due to the SDL version. Especially since it worked multiple times with a keyboard connected while it did not work without it in between.

christianhaitian, 2024-10-20:
So why were you able go update on a new sd card without issue?

TheColin21, 2024-10-20:
> So why were you able to update on a new sd card without issue?

That's the question 😅 I am also pretty sure it's not just due to the SDL version.... probably several factors combined. Or I could create some checksum. Maybe the file got corrupted (although that would also be surprising)

TheColin21, 2025-03-17:
This time I had to relink to the newest version 😅 3000.10