christianhaitian/arkos #1209: Wifi doesn't connect fast enough for retro achievements on quick mode

stiffx, 2024-10-24:
When using quick mode, Retroarch starts before the wifi connects so retro achievements does not connect. 

To successfully start retro achievements, you need to quit the game and open again. 

Running on:

RGB30 [v2.0.09.29.2024]

christianhaitian, 2024-11-01:
I don't have that issue when I test on my end but this has been mentioned by others.  The only way I know of currently to resolve this would be to add a delay during boot  before retroarch loads which somewhat defeats quick mode imo.

stiffx, 2024-11-05:
Could we have a not as quick mode, that just does the auto save on power off but boots to the menu?

RetroGFX, 2024-12-05:
> Could we have a not as quick mode, that just does the auto save on power off but boots to the menu?

I believe if using retroarch you can set it up to save your game on exit if that helps