christianhaitian/arkos #1210: Unable to update

Dantasstic, 2024-10-24:
### Describe the Issue (If applicable)

Unable to update to the latest ArkOS on my RG351V. 

Remote services are enabled, and Wi-Fi is up and working (_I can check and update RA cores_).
I tried disabling and re-enabling both, with no success.

I can enter OK, but then I get a screen that says:

> "There was an error attempting this update. Did you make sure to enable your wifi and connect to a wifi network? If so, enable remote services in options and try to update again."

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)

RG351V

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

ArkOS 2.0 (08232024)

christianhaitian, 2024-10-24:
Where are you located in the world?  Are you able to do other things using wifi like scraping your games for boxart or updating retroarch cores?

christianhaitian, 2024-10-24:
Have you been downloading shaders and other content through retroarch?

Dantasstic, 2024-10-24:
I haven’t recently, no, but I have done so in the past. I just updated the RA cores and core info files today. I also updated Portmaster with no issue today.

I don’t use the built in scraper, as I used the pc Skraper app to create my preferred game art.

christianhaitian, 2024-10-24:
Are you familiar with using ssh to terminal into ArkOS?

Dantasstic, 2024-10-25:
[image]

Got it ready to go

christianhaitian, 2024-10-25:
do `df -h` and post a snapshot of the result in this issue.

Dantasstic, 2024-10-25:
[image]

christianhaitian, 2024-10-25:
The linux partition is full.  That's why you can't update.  It may have happen after doing update GLSL Shaders and Overlays in Retroarch

Dantasstic, 2024-10-25:
Ok. How do I clear it?

christianhaitian, 2024-10-25:
Well I just checked and those shouldn't be the issue.  They are miniscule.  Have you been installing other packages using apt manually via ssh?

Dantasstic, 2024-10-25:
Nope, I only use what comes built-in with ArkOS, and only through the OS itself, never through SSH.

christianhaitian, 2024-10-25:
How long ago did you install the OS on this sd card?

Dantasstic, 2024-10-25:
Years ago. I've been updating no problem for a long time now. You think it just can't take being updated for that many iterations without a wipe and rebuild instead?

Dantasstic, 2024-10-25:
[image]

This current SD build is at least as old as this suggestion I had previously that got added, and for sure older than this

christianhaitian, 2024-10-25:
Ahh ok.  I had recently updated the initial OS sd card setup to allot more room for updates for the linux partition.  A new image install will be best so you can get the latest partition size update for the OS card.  You can use the settings backup and restore tool to help speed up the reinstall.  On a 32GB sd card for the OS, the linux partition now 11GBs instead of the 6.5GBs in the old setup process.

Dantasstic, 2024-10-25:
Ok, so I run dual cards. What is the proper procedure to start fresh but ensure I don't have to redo all my settings and layouts, etc.

christianhaitian, 2024-10-25:
Backup your current settings using the Backup ArkOS Settings in the Options>Advanced section
Install the new  image on your TF1 (OS) sd card
Restore your Settings
Update the OS

You may need to manually set your timezone as there was an issue with restoring that which was resolved in the 09/27/2024 update.

Dantasstic, 2024-10-25:
Ok, I have restored and updated to the latest build! I had to restore twice because it doesn't back up the themes from Thememaster, so I needed to update the themes first, and then restore again to get the theme back to where it was before. 

Other than that, it all seems to be as it was :) Thank you!

The newer build has created a 9.9GB partition, instead of the older 6.5GB one

closed 2024-10-25 by christianhaitian