christianhaitian/arkos #1213: Tate games

lauren7ino, 2024-10-26:
Hi
I've noticed that I can place Arcade roms is several folders, like: arcade, mame2003 and mame.

I use a Mame 2003+ non merged romset that belongs to the mame2003 folder.

These have a mix of horizontal and Tate games in there.

What would you do if you needed to create a separate folder only for Tate games?

Being in a separate folder is important because Tate games need a RetroArch screen tweak to change the proportions (and no rotation on the rgb30).

My current method implies a separate override for each Tate game, which is tedious.

Thank you in advance for any advice.

christianhaitian, 2024-10-26:
Hi,

You can edit `/etc/emulationstation/es_systems.cfg`, create a duplicate copy of the arcade system entry, have it point to a different folder name something like tate or vertical.  Some themes may have an entry for vertical only that you may be able to use with that system as well.  nes-box has a verticalarcade system.

lauren7ino, 2024-10-26:
It worked! Thanks! 👍🏼

closed 2024-10-27 by christianhaitian