christianhaitian/arkos #1214: Switch to SD2 for Roms - odd behavior / documentation bug

clach04, 2024-10-26:
### Describe the Issue (If applicable)

When using "Switch to SD2 for Roms" as described in https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RGB30#q-how-do-i-make-use-of-the-second-sd-slot-for-roms-and-games seeing a difference in behavior with docs (which was unexpected).

https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RGB30#q-how-do-i-make-use-of-the-second-sd-slot-for-roms-and-games states:
> This process will NOT copy any of your roms, games and saves from the main SD card to the secondary card. It will just create the structure for you. You'll need to copy those yourself....

This is not what happened ;-) The Ports directory is copied across, and I started with a single SD card and much later added/switched to a 2nd card and this process took a while.

Looking at https://github.com/christianhaitian/rk3326_core_builds/blob/rk3326/shell-scripts/351MP-V-only/Switch%20to%20SD2%20for%20Roms.sh#L104 this is clearly a deliberate decision :-)

### How can the issue be reproduced? (If applicable)

Take a single SD card, put a bunch of stuff in `/roms/ports`, then use `Switch to SD2 for Roms.sh`

### What device are you using?

RGB30

### What version of ArkOS are you on?

ArkOS version 2.0 (09292024) and ArkOS version 2.0 (05242024)

### Anything else you'd like to include that may help to review this issue or feature request?

**One** option would be to update the wiki to reflect this.

I want to help out. I have some ideas for refactoring (for simplification), I think it is possible to have the copying of ports (and maybe other directories) as an option.

I did start to refactor https://github.com/christianhaitian/rk3326_core_builds/blob/rk3326/shell-scripts/351MP-V-only/Switch%20to%20SD2%20for%20Roms.sh and then after doing that realized that the `Switch to SD2 for Roms.sh` under source code control is behind what I actually on the device 😿 . I have done some experimental modifications with the ArkOS version 2.0 (09292024) copy of `Switch to SD2 for Roms.sh` for rgb30 that worked for me BUT I suspect I should take a look at the upsteam version before doing any more testing.

Can you point me at the source code for `Switch to SD2 for Roms.sh`?

Thanks for making ArkOS available and continuing to work on it!

christianhaitian, 2024-10-26:
Oh you're right.  Not sure why I added that.  That latest version of the script will be available on the unit within the `/opt/systemAdvanced` section.  I"ve also updated the script via the link with the latest updates as well.

clach04, 2024-10-26:
@christianhaitian I _think_ change https://github.com/christianhaitian/rk3326_core_builds/commit/8183e496335fd9d73e0ac12fda9d7e2de7f16074 switches BACK to single sd card (like the https://github.com/christianhaitian/rk3326_core_builds/blob/rk3326/shell-scripts/351MP-V-only/Switch%20to%20main%20SD%20for%20Roms.sh script `Switch to main SD for Roms.sh`)

If I want to suggest changes, should I post PRs to https://github.com/christianhaitian/rk3326_core_builds? Would you prefer discussion in issues over there as well?

christianhaitian, 2024-11-01:
Discuss here is best

AeolusUX, 2024-11-02:
Why is this [new script](https://github.com/christianhaitian/rk3326_core_builds/blob/rk3326/shell-scripts/351MP-V-only/Switch%20to%20SD2%20for%20Roms.sh) linking roms instead of roms2? Isn't this the Switch to SD2 script?

> Discuss here is best

christianhaitian, 2025-01-05:
Any further updates on this?

clach04, 2025-01-11:
Sorry, I've not chance to look at this again (or play any games) 😔. I'm not sure when I'll get back to it.