christianhaitian/arkos #1216: Can we add an ons independent simulator

lcdyk0517, 2024-10-27:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANce FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

Independent emulator link：https://github.com/YuriSizuku/OnscripterYuri
This emulator supports *. nt3 games
Thanks

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2024-10-29:
There's a libretro build available for it in retroarch per https://github.com/YuriSizuku/OnscripterYuri/pull/30.  I've provided it in the repo.  You can try it by doing the following:  

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. If it ask if you want to delete the backup file, you can hit A for yes as you should not need it anymore.

4. Then proceed to test the core via the retroarch emulator

Let me know your results and if it should be included in a future update.

Thanks

[arkosbackup.tar.gz](https://github.com/user-attachments/files/17551682/arkosbackup.tar.gz)

lcdyk0517, 2024-10-29:
I tested several games on my RGB30, and Memories Off 2nd only had sound without any visuals. When testing the D.C.P.C game, the script for this game uses the nt3 format, and it cannot run with this core, causing it to crash immediately.

christianhaitian, 2024-10-29:
Do these games run on the standalone version of these emulators like for Windows?

lcdyk0517, 2024-10-29:
It has been tried and can run perfectly. It has also been tested with arm64 and there were no issues.

christianhaitian, 2024-11-01:
I got the standalone to build but I can't figure out what the expected controls are.  Is a mouse needed for most games?  Keyboard input doesn't seem to do hardly anything.  Any advice?

lcdyk0517, 2024-11-02:
Yes, most games are controlled with a mouse, but the keyboard is also needed to perform functions such as opening menus and fast-forwarding. I think it is possible to simulate the mouse with the left joystick and use the A button to simulate a mouse click. The ESC key to open the menu could be simulated with the B button.

christianhaitian, 2024-11-02:
I can't get a mouse pointer to appear yet.  So controlling the standalone is not good so far.

christianhaitian, 2024-12-04:
I'm still working on this and according to the emulator developer, a mouse cursor should be present so I'll investigate if it's an issue with the sdl library I'm using.

https://github.com/YuriSizuku/OnscripterYuri/issues/33

iyzsong, 2024-12-05:
You could try the latest build of onsyuri_libretro, which enable nt2,nt3 and LUA support, should fix D.C.P.C.
And Memories Off 2nd also works here, but after "Start Game" I have wait wait 5s and press OK or SKIP at least 2 times because in `11.txt`, it has:
 ```
bg "pic/notice.png",10,1000
wait 5000
click
click
```

lcdyk0517, 2024-12-05:
Can you share a compiled core?

iyzsong, 2024-12-05:
[onsyuri_libretro_aarch64_glibc2.38.zip](https://github.com/user-attachments/files/18017791/onsyuri_libretro_aarch64_glibc2.38.zip)
Here is a build for arm64 and glibc-2.38+, may or may not works on arkos, you could try or wait @christianhaitian to do a new build.

lcdyk0517, 2024-12-05:
thanks，I'll give it a try

christianhaitian, 2025-01-11:
I've update the libretro core in the repo

lcdyk0517, 2025-01-14:
I simply tested it and didn't find any bugs, and the Chinese ONS also runs well. I hope that in the next update, it can enter the game from the es frontend

christianhaitian, 2025-01-14:
So the libretro core seems sufficient now?

lcdyk0517, 2025-01-14:
At present, yes, the support for yurions is quite good. It would be even better if you could manually turn on/off the display/hide of the cursor

christianhaitian, 2025-02-01:
The onsyuri onscripter core has been added as an additional core for the onscripter system with today's update.

closed 2025-02-01 by christianhaitian