christianhaitian/arkos #1217: Fresh install for RG353M and hangs on ArkOS logo upon first boot

kevinandjenn123, 2024-10-28:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANce FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
Installed the latest RG353M ArkOS (from this hub)  .img to Samsung 512gb Pro Evo card, upon first boot just hangs on the ArkOS logo in the begining.  I unzipped image, wrote image with WinDisk32, IMGUSB, Pi-Imager all have the same effect.  NO I didnt use Balena.  

### How can the issue be reproduced? (If applicable)
Easily, everytime I turn on with card in, ArkOS logo pops up and hangs.  JELOS works tried that just to see, cant seem to get ArkOS to get past the ArkOS white logo upon first boot.

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)

RG353M - fresh from Amazon.

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

V2 09292024

### Anything else you'd like to include that may help to review this issue or feature request?
I can run JELOS from there Feburary 2024 build (last one, no more development) just fine on device.  Please help!!! ArkOS is what I want to use on here.  Has to be a fix, I cant be the only one with this issue?

christianhaitian, 2024-10-28:
You have a unit that has issues booting ArkOS stock.  See if this helps: https://github.com/christianhaitian/arkos/issues/837#issuecomment-2241738969

kevinandjenn123, 2024-10-28:
> (comment)

Thank you,

I'll give it a try tonight!!

christianhaitian, 2024-11-01:
Any update on this?

closed 2024-11-03 by christianhaitian

kevinandjenn123, 2024-11-03:
Works fine. Got the Patched file, thanks you