christianhaitian/arkos #1218: arkos-2.0.04.23.2022 Mirror?

johy-525, 2024-10-30:
hello，I want to get an image file,the version is arkos-2.0.04.23.2022

christianhaitian, 2024-10-30:
I do not have this image available. It’s much to old.  Why would you want this old of a version?

johy-525, 2024-10-30:
I bought a game console and asked for this version, which looks more stable.
This image can be DIY?, how to compile and generate?

christianhaitian, 2024-10-31:
What console is this?

johy-525, 2024-11-01:
R36s

christianhaitian, 2024-11-01:
You can find a community maintained version of ArkOS for the R36s here: https://aeolusux.github.io/ArkOS-R3XS/
This OS is not easy to build as it's a mostly manual process.  You can see here for the documentation.  https://github.com/christianhaitian/arkos/wiki/Building
As I don't support the R36s units, I don't know how best to advise you then to probably get on the RH Discord and find the gameconsole r36s channel there for better advice.  Good luck.

closed 2024-11-01 by christianhaitian