christianhaitian/arkos #1219: Add support for PowKiddy RGB20 Pro

RetroBreeze, 2024-11-02:
### Describe the Issue (If applicable)
Can you add support for the PowKiddy RGB20 Pro? It uses the RK3566 chipset along with a 3.2" 1024*768 screen.

christianhaitian, 2024-11-02:
I may add that unit.  I'm just waiting for my unit to arrive.

okiookio, 2024-11-07:
> I may add that unit. I'm just waiting for my unit to arrive.

When did you place your order?
I’m also waiting like you and hope that when the device arrives, I’ll be able to install updates from you.
Thanks

christianhaitian, 2024-11-10:
Just got mine last week.  Haven’t spent much time with it yet as I’m spending time with family at the moment.

RetroBreeze, 2024-11-10:
> I think about 2 to 3 weeks before your request here. Just got mine last week. Haven’t spent much time with it yet as I’m spending time with family at the moment.

Thanks for the update! Looking forward to it, but please take your time.

christianhaitian, 2024-11-19:
I have a build of ArkOS running on the RGB20 Pro.  It does require that the unit be opened and the battery unplugged for 10 seconds or it will not be able to safely shutdown.  It's been a known issue when running a mainline kernel based OS like ROCKNIX then switching to a BSP Kernel based OS like ArkOS.

okiookio, 2024-11-20:
My RGB20Pro hasn’t been delivered yet, but I am currently using both the RGB30 and RGB20SX. This issue only occurs on the RGB30 and does not affect the RGB20SX at all. My RGB20SX is from the latest batch manufactured in 2024. I’m quite surprised by this as well. Hopefully, this information will be helpful to you.

christianhaitian, 2024-11-28:
RGB20 Pro support is now available.  You can download a compatible image from the Download links section of the wiki.  It will be labelled specifically for the Rgb20 Pro  

**Important Note!**
If your unit came with ROCKKNIX installed or you've run ROCKNIX on your units, you must open your unit and unplug the battery for 10 seconds or it will not be able to safely shutdown.  The unit will just reboot.
As a temporary workaround, if you shutdown the unit in ROCKNIX using the the proper shutdown process through the start menu, you will be able to shutdown safely in ArkOS, however, the solution is temporary because if you do a system reboot, the issue will return.

closed 2024-11-29 by christianhaitian