christianhaitian/arkos #1220: Request: Scraper To Update Metadata Only When Field Values are Invalid (No file at path) or Blank, Or Update Only Selected Fields

so33-bit, 2024-11-04:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANce FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

Not a preloaded image.

3. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

Yes

4. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

n/a

6. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

Yes

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

I have consistent metadata issues for almost all games (not the fault of arkos). Some of the same fields are blank (e.g. marquee) or have file paths that are invalid ('video' has no file at that location). If I scrape for 'missing' metadata, all the data is updated. As was the case of another issue I found here since closed, my gamelist.xml files might not be quite right. 

I would just scrape it all, but I don't want to have my images be overwritten; I prefer my existing images to those at the scraper site.

My request is that the scraper optionally update field values when they are either blank or consist of a invalid/missing file-path only. Alternatively, I request to have an option to select which metadata fields to scrape for (that way, I could de-select 'image').

### How can the issue be reproduced? (If applicable)

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)

RGB30

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

20241025

### Anything else you'd like to include that may help to review this issue or feature request?

so33-bit, 2024-11-04:
It would also solve my issue if the Scraper settings were such that 'Image Source' allowed 'none', if that would skip updating the images during scraping.

christianhaitian, 2024-11-05:
I'll see what is possible.

christianhaitian, 2024-11-05:
See if this works for you:

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1(TF1 Slot).  

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, reboot the device.

You should now be able to set image source to none and scrape.  Let me know if this achieves what you're looking for.

Thanks,

[arkosbackup.tar.gz](https://github.com/user-attachments/files/17625825/arkosbackup.tar.gz)

so33-bit, 2024-11-05:
Thank you, but somehow this ends up deleting the paths to image, video, marquee, and thumbnail after scraping.

christianhaitian, 2024-11-05:
Interesting.  It deletes the paths but doesn't delete the actual media correct?

so33-bit, 2024-11-05:
That's correct, but it never deleted or overwrote the actual media, as the file-paths were different. My images are in roms2/{system}/media/images. Also, the file names are usually a little different from what is scraped.

christianhaitian, 2024-11-19:
I thought this would be simpler to figure out than it has been.  Not sure this is something I will be able to figure out if it's feasible or not.

so33-bit, 2024-11-19:
I'm a developer, but I don't know C++.. I wish I could contribute better.

I feel that the issue is here: https://github.com/christianhaitian/EmulationStation-fcamod/blob/ce0fa043e2a1dbd6cc10fb2bcce57bd6028b12d9/es-app/src/scrapers/Scraper.cpp#L218

I would expect that this would not overwrite an image already specified in the metadata; it looks like it should work as-is.

I just tried it again, and the Scraper is not working at all now. Could it be the [arkosbackup.tar.gz](https://github.com/user-attachments/files/17625825/arkosbackup.tar.gz) that I applied? I changed the value for "Image" from "None" to "Box 2D", but the Scraper gets stuck at downloading now.

christianhaitian, 2024-11-19:
Are you sure your network connection is working?  Are you able to ping the device while on the network?  screenscraper could have been also having issues earlier.  Maybe try it again?

so33-bit, 2024-11-19:
Ok! The "Scrape Now" function wasn't working with "User decided on conflicts" turned ON. After turning it off, the scraping worked exactly as I hoped, the "Only missing medias" option prevented overwriting of my images.

I see that the problem I'm facing might be limited to scraping within the "Edit this Game's Metadata' function; that one will overwrite existing media file-paths. Scraping a collection entirely does work as I hoped.

Would it be feasible to add a "Only missing medias" option to scraping for a single game using "Edit this Game's Metadata' function?

christianhaitian, 2024-11-20:
I'm sure it is but I"m not sure how best to implement such a feature.

christianhaitian, 2024-11-29:
Closing this issue now as this is not something I will further work on.

closed 2024-11-29 by christianhaitian