christianhaitian/arkos #1221: Add support for X55

NeveIsa, 2024-11-05:
Since RGB30 and X55 have same hardware except for display, could we please make support for X55?

My unit is arriving in a week and I would like to learn and help about the building of Linux for this device. 

Regards,
Sampad

christianhaitian, 2024-11-05:
Hello,

The X55 is not a unit I'm interested in supporting.

Thanks

closed 2024-11-05 by christianhaitian

NeveIsa, 2024-11-05:
Could you help me get started on it , like any tips on where to look for and how to extract the dtb and stuff from the original factory firmware?

christianhaitian, 2024-11-06:
The dtb is probably on the OS card on the unit.  On a Ubuntu machine, make sure to install device-tree-compiler from the apt repo.  Then do
`dtc -I dtb -O dts devicetree_file_name.dtb > devicetree_file_name.dts`

Make the panel changes then you can recompile the dtb using the following command:
`dtc -I dts -O dtb -f devicetree_file_name.dts -o devicetree_file_name.dtb `

There is also [ROCKNIX](https://rocknix.org/) available for the X55 and my understanding is that it works very well for the X55.  Good Luck!