christianhaitian/arkos #1224: Custom picture for slideshow screensaver has wrong orientation

ridhoperdana, 2024-11-19:
### How can the issue be reproduced? (If applicable)
1. Copy picture files with format .jpg and .png to sd card (roms2/slideshow)
2. Activete slideshow screensaver
3. The picture shown at screensave has wrong orientation (all my picture randomly rotated)
 3a. If i check the picture on my phone or PC (i am using mac and ios) the picture orientation is correct.

### What device are you using? 
RGB30

### What version of ArkOS are you on?
10252024

christianhaitian, 2024-11-19:
Can you provide a sample of the image here so I can test on my end?  I just tested on my end and the images are not rotated on my end as shown in the attached pictures below.

[image]
[image]

ridhoperdana, 2024-11-19:
I am afraid I can't, it is my family photo so I can make my RGB30 as photoframe. Can you guess why is it happen to my image? is it because the resolution?

ridhoperdana, 2024-11-19:
I just fixed it by using another picture, maybe the first picture i use has wrong orientation and somehow my phone and PC automatically fixed it.

Closing this, thanks!

closed 2024-11-19 by ridhoperdana