christianhaitian/arkos #1225: ScummVM not remembering options

kleingeld1, 2024-11-24:
Not on preloaded images, yes I checked the FAQs. 

### Describe the Issue (If applicable)
ScummVM does not remember the settings for volume and subtitles. Upon next start all volume settings are back to 255 and subtitles off. Tried different games (samnmax, brokensword). 

### How can the issue be reproduced? (If applicable)
Set volume and or subtitles. Save game. Restart ScummVm. 

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)
RGB30 and R36S

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
2.0 (09292024)

### Anything else you'd like to include that may help to review this issue or feature request?
—

christianhaitian, 2024-11-25:
Are you using a 2 SD card setup or just 1?

christianhaitian, 2024-11-25:
Try doing Switch to main sd card, reboot the unit, then do switch to sd2 again and see if that resolves the issue for you.

kleingeld1, 2024-11-25:
I tried that, but the issue still persists.

christianhaitian, 2024-11-25:
You have room left on your 2nd sd card?

kleingeld1, 2024-11-25:
53,35 GB of 62,5 GB available

christianhaitian, 2024-11-25:
Odd.  The only way I could replicate this issue is if the save was trying to happen to a location that doesn't exist or if the disk was full or locked. Did you happen to change the location for saves away from the default location (which for a 2 sd card setup would be /roms2/scummvm)?

kleingeld1, 2024-11-26:
It‘s exFAT. I haven’t changed the save path and in-game saving works flawlessly.

kleingeld1, 2024-11-26:
Yep. Please note that this is my RGB30's output only, I can't do it on my R36S as I don't have a Wifi dongle for it (so the "free" numbers will differ from my last post, but both are using exFAT).
`ark@rgb30:~$ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            470M     0  470M   0% /dev
tmpfs            97M  2.3M   95M   3% /run
/dev/mmcblk1p4  9.9G  6.5G  3.0G  69% /
tmpfs           483M     0  483M   0% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           483M     0  483M   0% /sys/fs/cgroup
/dev/mmcblk1p3   99M   32M   68M  32% /boot
/dev/mmcblk2p1  120G   59G   61G  49% /roms2
/dev/mmcblk1p5   50G  535M   49G   2% /roms
tmpfs            97M     0   97M   0% /run/user/1002`

christianhaitian, 2024-11-26:
Thank you.  I don't support the r36s so the rgb30 is good.  Try running the following command via ssh and see if it makes a difference with saving for you.

`sudo chown -R ark:ark /home/ark`

This will make sure that all files and folder within the ark subfolder are owned by the ark user.

kleingeld1, 2024-11-26:
Found a way to df -h on my R36S as well:
[image]

christianhaitian, 2024-11-26:
Did you try this yet?

`sudo chown -R ark:ark /home/ark`

christianhaitian, 2024-11-26:
How are you quitting ScummVM?

kleingeld1, 2024-11-26:
Start to open Menu -> Quit ScummVM

christianhaitian, 2024-11-26:
Unfortunately, I'm out of ideas on what the issue can be then.

christianhaitian, 2024-11-26:
I don't think so.  You can try checking the `/home/ark/.config/scummvm/scummvm.ini` file and verify the save location is set to `/roms2/scummvm`

kleingeld1, 2024-11-26:
I found a workaround. Starting just ScummVM and adjusting the global options persists across restarts. Not ideal, but works for now.

christianhaitian, 2024-11-29:
Ok.  I can't replicate this exact issue unfortunately.

closed 2024-11-29 by christianhaitian

Any-berg, 2025-04-17:
@kleingeld1 hey, if you are still looking for a better solution, you might want to check mine in a related issue. Testing it requires you to ssh into your device and sudo edit `/usr/local/bin/scummvm.sh`.