christianhaitian/arkos #1228: Power Button issue after update.

snakkarike, 2024-11-29:
I recently updated to the 29th November build and there has been a issue with the power button. It doesnt register power button correctly to lock the screen and have to press a few times. Upon pressing and turning on the screen. A volume bar like display pops up for a few seconds which seems to show 24% on the right side. The volume percentage is on the left and that is working fine.
[image]

christianhaitian, 2024-11-29:
Sounds like possible corruption with the sd card.  What brand sd card are you using?  If Sandisk or Samsung, Have you tried a fresh install yet?

snakkarike, 2024-11-29:
> Sounds like possible corruption with the sd card. What brand sd card are you using? If Sandisk or Samsung, Have you tried a fresh install yet?

I thought so too. so i formatted the sd card, checked for any errors in the card and reflashed ark os and did a clean install. The image i attached was after the clean install. Had issues after the update where the system would lag and recover when scrolling through the games and the power button issue so i did do a clean install the lag is not there for now but the power button issue is there.

snakkarike, 2024-11-29:
Since it is an RG353V. I checked on the Android OS to rule out hardware related issue. The power button works fine on Android.

christianhaitian, 2024-11-29:
What brand sd card are you using?

christianhaitian, 2024-11-29:
This really sounds like a bad sd card

snakkarike, 2024-11-29:
Sandisk extreme 64gb v30 A2

christianhaitian, 2024-11-29:
Hmm, I use this card regularly and don't experience this issue.  What happens when you load a fresh image on the card and use it as is before updating the image?  Also, what software are you using to image the sd card?

snakkarike, 2024-11-29:
I will try a fresh install without an update and check. I used rufus to flash the image.

RetroGFX, 2024-11-29:
> I will try a fresh install without an update and check. I used rufus to flash the image.

I've had weird issues with Rufus and etcher.. try win32diskimager see if that helps. Also someone's a quick format isn't the way to go as it doesn't zero out sectors or blocks. Try a full format if win32 doesn't fix it

christianhaitian, 2024-11-30:
I think I was able to replicate a stability issue and believe it's related to the updated exfat kernel module.  Attached is a zip file and within it is a script that you can put into the tools folder of your games card and run it from Options>Tools section in Emulationstation.  It will be named **Exfat_Module_Revert**.  Just press A to launch the script, and Press A to continue.  Once it's completed, the unit will reboot.  Hopefully that should fix the lag and power button issue for you.  If it helps or not, please respond here so I can plan to include this fix in the next update if need be.

[Exfat_Module_Revert.zip](https://github.com/user-attachments/files/17963567/Exfat_Module_Revert.zip)

snakkarike, 2024-11-30:
I ran the script and the system rebooted but the issue still persists. I will clean flash the older arkos image and check.

snakkarike, 2024-11-30:
I flashed the older arkos image. 09292024 it has the same power button issue i dont know how to figure out the issue.

snakkarike, 2024-11-30:
I loadeds the sdcard in mini partition wizard. Wiped the card checked the sectors cleaned it and then flashed arkos using win32diskimager. The issue still persists. I checked on the android again to rule hardware issue and the button works fine with no delay. There is a significant delay in ArkOS in recognizing the button press. I am currently on 09292024 build.

snakkarike, 2024-11-30:
Okay upon further inspection i figured that the 24% that is being shown is the brightness or BRT. In the settings i have the info of Battery, SND and BRT. BRT is the only thing which is at 24% and what is currently being displayed on pressing the power button. Is there a way to disable it? And i think the power button latency is coming from the script that puts the device to sleep probably.As pressing the power button after like 10 seconds makes it work somewhat better.
[image]

christianhaitian, 2024-11-30:
Oh yes, when you wake the device from sleep, the brightness is reset to where it was when initially put to sleep.  That will show in the ES menu for a brief moment upon wake.  That’s normal. Also, upon wake, there will be a small period of time while various features are reenabled where global hotkeys will not work.  You can shorten that by disabling bluetooth as that will cause a longer delay for the hotkeys to reenable.

i was under the impression that this was nit normal in your regular usage before the update but from what you describe now, that is the expected behavior.

snakkarike, 2024-11-30:
Ah okay. Thank You sorry for bothering. But reflashing fixed the stability issues so that is good.

christianhaitian, 2024-11-30:
No worries. Also, the lag upon wake is probably because prior to sleep, the unit goes to the powersave cpu governor to further reduce power consumption while sleep.  Upon wake, it puts the cpu back to whatever governor it was last at which can take some seconds.  That can be sped up by disabling bluetooth if you’re not using that.

take care.

closed 2024-11-30 by christianhaitian

Plaque-fcc, 2025-04-13:
After copying directories from the EASYROMS partition of an ancient install, this issue happened to me as well.
Will be careful at copying them here and there.