christianhaitian/arkos #1229: bug - ArkOS 09292024 for PowKitty RGBMax10 - Drastic hotkey Issue

dremond71, 2024-11-30:
### Describe the Issue (If applicable)

On Powkitty RGB10x , 

[image]

which has 1 analog stick.

The hotkey to bring up the Drastic emulator menu is set to the same keycode as the analog stick 'press down' keycode. (stylus click keycode). So this means, in a game which depends on a stylus click, when the user does this, it brings up the Drastic emulator menu....instead of clicking down on the on the graphical item below the stylus.

### How can the issue be reproduced? (If applicable)

Open a nintendo ds game, Super Princess Peach (for example), move the stylus over one of the 3 heart icons, and press down on the analog joystick. It will open the Drastic emulator menu.

### What device are you using?

Powkitty RGB10x

### What version of ArkOS are you on?

Arkos 09292024

### Anything else you'd like to include that may help to review this issue or feature request?

I changed my Powkitty RGB10x's

/opt/drastic/config.cfg to this:

```
frameskip_type = 2
frameskip_value = 1
safe_frameskip = 0
show_frame_counter = 0
screen_orientation = 2
screen_swap = 0
savestate_number = 0
fast_forward = 0
enable_sound = 1
clock_speed = 0
threaded_3d = 1
mirror_touch = 0
compress_savestates = 1
savestate_snapshot = 1
unzip_roms = 0
preload_roms = 0
backup_in_savestates = 1
ignore_gamecard_limit = 0
frame_interval = 0
trim_roms = 0
fix_main_2d_screen = 0
disable_edge_marking = 0
interframe_blend = 0
hires_3d = 0
use_rtc_custom_time = 0
rtc_custom_time = 0
rtc_system_time = 0
slot2_device_type = 0
rumble_frames = 3
firmware.username = Dr DraStic
firmware.language = 1
firmware.favorite_color = 0
firmware.birthday_month = 1
firmware.birthday_day = 1
enable_cheats = 1
controls_a[CONTROL_INDEX_UP] = 1217
controls_a[CONTROL_INDEX_DOWN] = 1153
controls_a[CONTROL_INDEX_LEFT] = 1216
controls_a[CONTROL_INDEX_RIGHT] = 1152
controls_a[CONTROL_INDEX_A] = 1025
controls_a[CONTROL_INDEX_B] = 1024
controls_a[CONTROL_INDEX_X] = 1026
controls_a[CONTROL_INDEX_Y] = 1027
controls_a[CONTROL_INDEX_L] = 1028
controls_a[CONTROL_INDEX_R] = 1029
controls_a[CONTROL_INDEX_START] = 1037
controls_a[CONTROL_INDEX_SELECT] = 1036
controls_a[CONTROL_INDEX_HINGE] = 104
controls_a[CONTROL_INDEX_TOUCH_CURSOR_UP] = 1219
controls_a[CONTROL_INDEX_TOUCH_CURSOR_DOWN] = 1155
controls_a[CONTROL_INDEX_TOUCH_CURSOR_LEFT] = 1218
controls_a[CONTROL_INDEX_TOUCH_CURSOR_RIGHT] = 1154
controls_a[CONTROL_INDEX_TOUCH_CURSOR_PRESS] = 1038
controls_a[CONTROL_INDEX_MENU] = 109
controls_a[CONTROL_INDEX_SAVE_STATE] = 318
controls_a[CONTROL_INDEX_LOAD_STATE] = 320
controls_a[CONTROL_INDEX_FAST_FORWARD] = 8
controls_a[CONTROL_INDEX_SWAP_SCREENS] = 115
controls_a[CONTROL_INDEX_SWAP_ORIENTATION_A] = 97
controls_a[CONTROL_INDEX_SWAP_ORIENTATION_B] = 100
controls_a[CONTROL_INDEX_LOAD_GAME] = 65535
controls_a[CONTROL_INDEX_QUIT] = 65535
controls_a[CONTROL_INDEX_FAKE_MICROPHONE] = 65535
controls_a[CONTROL_INDEX_UI_UP] = 338
controls_a[CONTROL_INDEX_UI_DOWN] = 337
controls_a[CONTROL_INDEX_UI_LEFT] = 336
controls_a[CONTROL_INDEX_UI_RIGHT] = 335
controls_a[CONTROL_INDEX_UI_SELECT] = 13
controls_a[CONTROL_INDEX_UI_BACK] = 8
controls_a[CONTROL_INDEX_UI_EXIT] = 27
controls_a[CONTROL_INDEX_UI_PAGE_UP] = 331
controls_a[CONTROL_INDEX_UI_PAGE_DOWN] = 334
controls_a[CONTROL_INDEX_UI_SWITCH] = 481
controls_b[CONTROL_INDEX_UP] = 1032
controls_b[CONTROL_INDEX_DOWN] = 1033
controls_b[CONTROL_INDEX_LEFT] = 1034
controls_b[CONTROL_INDEX_RIGHT] = 1035
controls_b[CONTROL_INDEX_A] = 65535
controls_b[CONTROL_INDEX_B] = 65535
controls_b[CONTROL_INDEX_X] = 65535
controls_b[CONTROL_INDEX_Y] = 65535
controls_b[CONTROL_INDEX_L] = 65535
controls_b[CONTROL_INDEX_R] = 65535
controls_b[CONTROL_INDEX_START] = 13
controls_b[CONTROL_INDEX_SELECT] = 538
controls_b[CONTROL_INDEX_HINGE] = 65535
controls_b[CONTROL_INDEX_TOUCH_CURSOR_UP] = 65535
controls_b[CONTROL_INDEX_TOUCH_CURSOR_DOWN] = 65535
controls_b[CONTROL_INDEX_TOUCH_CURSOR_LEFT] = 65535
controls_b[CONTROL_INDEX_TOUCH_CURSOR_RIGHT] = 65535
controls_b[CONTROL_INDEX_TOUCH_CURSOR_PRESS] = 1038
controls_b[CONTROL_INDEX_MENU] = 1039
controls_b[CONTROL_INDEX_SAVE_STATE] = 65535
controls_b[CONTROL_INDEX_LOAD_STATE] = 65535
controls_b[CONTROL_INDEX_FAST_FORWARD] = 65535
controls_b[CONTROL_INDEX_SWAP_SCREENS] = 1040
controls_b[CONTROL_INDEX_SWAP_ORIENTATION_A] = 1031
controls_b[CONTROL_INDEX_SWAP_ORIENTATION_B] = 1030
controls_b[CONTROL_INDEX_LOAD_GAME] = 65535
controls_b[CONTROL_INDEX_QUIT] = 65535
controls_b[CONTROL_INDEX_FAKE_MICROPHONE] = 65535
controls_b[CONTROL_INDEX_UI_UP] = 1032
controls_b[CONTROL_INDEX_UI_DOWN] = 1033
controls_b[CONTROL_INDEX_UI_LEFT] = 1034
controls_b[CONTROL_INDEX_UI_RIGHT] = 1035
controls_b[CONTROL_INDEX_UI_SELECT] = 1025
controls_b[CONTROL_INDEX_UI_BACK] = 1026
controls_b[CONTROL_INDEX_UI_EXIT] = 1024
controls_b[CONTROL_INDEX_UI_PAGE_UP] = 1028
controls_b[CONTROL_INDEX_UI_PAGE_DOWN] = 1029
controls_b[CONTROL_INDEX_UI_SWITCH] = 1039

```

Essentially

I changed analog stick press to 1038

```
controls_a[CONTROL_INDEX_TOUCH_CURSOR_PRESS] = 1038
controls_b[CONTROL_INDEX_TOUCH_CURSOR_PRESS] = 1038
```
and I changed

these fields to 1039 ( code for pressing on '+' button on device)
```
controls_b[CONTROL_INDEX_MENU] = 1039
controls_b[CONTROL_INDEX_UI_SWITCH] = 1039
```

So now, on the device : 

- any time I want to bring up the drastic emulator menu, I press on '+'
- any time I want to 'click the stylus', I press down on the analog stick

dremond71, 2024-11-30:
Besides this tiny issue, I find the ArkOS system to be awesome. Thx for creating and maintaining this. :)

christianhaitian, 2024-11-30:
Thanks for the settings recommendation.  I knew that was an issue but the same image used for the rgb10x is primarily focused on the rg351mp and I didn't take the time to figure out how to best address this for the rgb10max.  I'll keep this configuration handy for possible future inclusion in a future update.  Enjoy!

dremond71, 2024-11-30:
Cool. You are welcome :)

dremond71, 2024-12-01:
Cool.

Maybe put it in the readme or wiki
under RGB10x category or something?
Just a thought. :)

GuiGayah, 2025-02-08:
I have an rgb10x and I would like to know how to update its system, I couldn't find the updated Arkos for it, if anyone has it, can you send me the link please

christianhaitian, 2025-02-08:
Use the RG351MP image then apply additional updates via wifi

jonathan17819, 2025-02-20:
Add the more recent nds (drastic) emulator