christianhaitian/arkos #1230: Achievement progress lost

Qubits01, 2024-12-01:
Hi, 

using the latest version on a rg353m.
It seems like the wifi disconnected while playing. Achievements continued to get displayed when achieved but were not registered on the size. When realized that, I tries to reconnect wifi from the ingame menu (select+x). a box with "searching for wifi" appeared but stayed there forever. Eventually I tried to exit the game (select+start) and there it would not return to ES. The system hung up and was unresponsive, so I had to force shutdown.

MY question: What's the expected behavior when wifi gets lost while playing? Could my problem even be hardware related? The system hanging up made me wonder. Any hints how to probleshoot further?

christianhaitian, 2024-12-01:
I've never experienced this issue so I'm not sure honestly.  What version of ArkOS are you on?

Qubits01, 2024-12-01:
I'm on 09292024 (corrected above).

What would the behavior be when you'd play with achievements and let's say walk out of wifi range or switch off your wifi router?

christianhaitian, 2024-12-01:
maybe just disconnect and things continue to function just without achievements working is my guess.

christianhaitian, 2024-12-04:
Any update on this?  Have you experienced this issue again?

Qubits01, 2024-12-09:
Hi, sorry for the late reply.
I did an update to the latest of arkos. I had an instance where apparently the connection got lost and I got an on screen notification that the connection to RA server got lost, trying to reconnect nad syncing the progress. So that's what's supposed to happen.

About your question: I was not able to play much recently (hence the late answer) but I had once instance recently (with the latest version) where again the system would hang if I try to exit the game and return to arkos (2x start+select).
I did not unlock any achievements in that session but the symptoms were the same as initially described.
Is there any type of system logging to find out what's going on with the wifi connection?

christianhaitian, 2024-12-10:
What brand sd card are you using and what software did you use to flash the OS onto the card?

Qubits01, 2024-12-10:
SD card brand is a sandisk (32GB) If it's relevant I can look up the specific variant. I used Rufus to flash. I also have a Miyoo A30 with spruce where I use the same kind of card and also used rufus and I did not encounter any problems with that device so far.

christianhaitian, 2024-12-10:
When it hangs, did you wait anytime to see if the system would eventually respond again?

Qubits01, 2024-12-10:
I waited several minutes. If you think it would make sense, I can wait longer next time this occurs.

christianhaitian, 2024-12-10:
No that shouldn't be needed at all.  What happens if you disable wifi and play without it for awhile?

Qubits01, 2024-12-10:
I can try that. Problem is that I can not deterministically reproduce the issue. There are sessions, also pretty long ones where I am able to play with wifi without the issue occuring. So it will be hard to "prove" that the issue will never happen without wifi.
However, the issue I described in the OP (status message not disappearing when trying to reconnect to wifi) is something that shouldn't happen as well and most probable wifi related.

To ask again: Is there any way to get linux system based infos, usually in form of a log file, to get information about the state of the wifi connection when the problem occurs. For example why the disconnects happen. I think that would be most useful.

christianhaitian, 2024-12-10:
Not by default.  You'd have to enable logging in the retroarch menu.

christianhaitian, 2025-01-02:
Any additional updates on this?  Have you experienced this issue again?

christianhaitian, 2025-01-05:
Closing due to lack of response and ability to replicate the issue

closed 2025-01-05 by christianhaitian