christianhaitian/arkos #1231: Upstream `gamecontrollerdb.txt` mappings

offalynne, 2024-12-05:
Is there a compelling reason `gamecontrollerdb.txt` changes are not upstreamed to the source repo https://github.com/mdqinc/SDL_GameControllerDB ?

I suppose if there are udev drivers or other changes that make Ark different than a generic distro this might cause compatibility conflict but surely this is not the case for all handheld device controls ? I think it would be ideal for as many mappings as possible to move upstream

christianhaitian, 2024-12-05:
I tried this before and the result wasn't what I expected <https://github.com/mdqinc/SDL_GameControllerDB/pull/622>.  It's your project and you're fine to do with it as you want but it also means I have no longer have interest in trying to assist with this either.  Thanks

closed 2024-12-05 by christianhaitian

offalynne, 2024-12-06:
Hi Christian,

As explained in the PR you linked, the description field in SDL mappings is supposed to resemble what one would find on consumer-facing marketing rather than a direct copy of what is found in a HID report device description, for example "PS5 Controller" as opposed to "Sony Interactive Entertainment Wireless Controller". 

I'm confused as to why this semantic difference has you determined not to contribute to the community database you are actively using yourself, but so it goes, I guess.