christianhaitian/arkos #1232: Kid Mode Blocks Games from Some Systems

rbaird41, 2024-12-07:
When in kid mode, my game lists for TurboGrafx 16, TurboGrafx-CD, Dreamcast, Sega CD, and a couple others are empty. Does Kid mode block certain systems? If so, can that be changed somehow? Sorry if this has been answered, but I didn't see any open or closed issues for this particular thing. Thanks in advance.

christianhaitian, 2024-12-07:
Hello,

The expectation is that you've marked the games you want available as kidgame before you enable kid mode..  You do this by scrolling until the game you want to tag as a kidgame, press Select, then press A on **Edit This Game's Metadata**.  Scroll until **Kidgame** and toggle to **On** by pressing A on it..

rbaird41, 2024-12-07:
That'll take a while, but I'll work on it. Thanks!

closed 2024-12-07 by christianhaitian