christianhaitian/arkos #1234: rgb30

retrooldman, 2024-12-12:
RGB30 upgrade to the latest version of the game vibration function is invalid,e.g.:GBA (Screw Breaker-Goushin Dorirurero
0, but PS Tekken III is normal .Is it a problem with the RetroArch version?             

（Version 0823 arkos Everything is OK）

christianhaitian, 2024-12-12:
Did you update the cores in retroarch?  If so, maybe the mgba rumble is not functioning.  I'll check this out.

christianhaitian, 2024-12-13:
I found the issue.  It was related to the rumble patch not being correctly updated back in July.  This has been resolved and the core has been updated.  Just do a core update in retroarch and that should resolve it for you.  Please complete this and provide an update on if this issue is now resolved for you or not.

retrooldman, 2024-12-13:
Thanks for the reply, but I don't know how to update this fix. After updating the mgba core on RetroArch, it still won't vibrate。How should I fix it? Thank you.

christianhaitian, 2024-12-13:
I just updated the mgba_rumble core via retroarch on my end and the mgba_rumble core works fine on my end.  Did you make sure that you were connected to wifi, loaded retroarch directly and do update installed cores?

retrooldman, 2024-12-14:
[image]
[image]
[image]
[image]
[image]
Yes, I have updated the core of retroarch and mgba to the latest version, and the vibration in the game settings is turned on, but the vibration still does not work.

christianhaitian, 2024-12-14:
Ok.  I was able to replicate the issue on a fresh install.  I've reverted the rumble patch to the previous version that worked from 0823.  Seems to work for me.  Please confirm it works for you as well.

retrooldman, 2024-12-15:
Nice, now I updated the installed core in retroarch and the Pokémon pinball，Drill Dozer is vibrating, thank you very much for your prompt reply, my handheld is reborn，thanks again

christianhaitian, 2024-12-15:
Awesome!  Thanks for confirming.  Enjoy!