christianhaitian/arkos #1238: Only PlayStation showing on systems list in Kid mode

rbaird41, 2024-12-25:
I'm on the most recent release of Arkos, and booting in Kid mode only shows PlayStation on the systems list. If I go to all games, I can see every game from every system that I marked as being for kids. Once I switch to the full UI and restart, all of the systems show. Then they remain there when I switch to Kid mode. But after I restart, they're gone again. Is there a setting somewhere that's making this happen? Thanks for your help!

christianhaitian, 2024-12-25:
Merry Christmas to you as well.  What unit are you using? Are you using a good quality sd card such as Sandisk or Samsung? How are you shutting down the system?

rbaird41, 2024-12-25:
It's happening on both of my RGB30 devices. I'm using SanDisk cards, and I
have it set up to shut down with just the power button so it's easier for
my kids.

christianhaitian, 2024-12-25:
Hmm...that should be fine then.  Let me see if I can replicate this issue on my end.

christianhaitian, 2024-12-25:
I was able to replicate the issue.  Looks like when marking games as kid friendly the proceeding immediately to KId mode is not allowing those changes to save in the gameslist.xml file during shutdown as should be happening.  Workaround for the issue for now is to just restart emulationstation through the start>quit menu after setting your games to kid mode.  Then place the unit into kid mode and the games should remain even when using the shutdown hotkey.  Thank you for reporting the issue.

rbaird41, 2024-12-25:
Are you saying I'd have to clear all of the kid game settings and set them
all as kid games again? And then restart emulationstation before switching
to kid mode? I have something like 600+ games marked as kid games, so I'd
rather not have to do that. In the meantime, I can just keep exiting kid
mode, restarting emulationstation, and then putting it back into kid mode
before they play. At least until you're able to come up with a fix. Maybe
I'm not following exactly what you're suggesting.

christianhaitian, 2024-12-25:
No, you don't have to clear any kid game settings.  After you've marked any additional games for kid friendly, restart emulationstation.  Then enable kid mode.

rbaird41, 2024-12-25:
That works fine until I shut down the device. But when I restart it after
it's already in kid mode, PlayStation is the only system showing again.

christianhaitian, 2024-12-25:
Double check and verify that the other games are still marked as kid friendly.  It sounds like they may not be.

rbaird41, 2024-12-25:
I just checked several from the systems that are missing in kid mode, and
they're all still marked as kid games. And all of the kid games show in kid
mode when I go to my All folder. But the only system that shows in the menu
is PlayStation.

christianhaitian, 2024-12-25:
Hmm....  Let me see if I can replicate this.

christianhaitian, 2024-12-25:
I can't replicate an issue of just playstation showing when kid mode is enabled but various other system games are marked as kid game.  Did you change any other settings in the start menu other than enable all games in the automatic game collections?

rbaird41, 2024-12-25:
I have the systems sorted alphabetically, but switching that back to "no"
didn't make any difference. I don't recall changing any other settings. All
of the systems are set to be visible.

christianhaitian, 2024-12-25:
Unfortunately, I'm at a loss on this one as I simply can't make only playstation show in the menu on my end with kid mode enabled.  A matter of fact, every system with games loaded still shows, they just don't have any game entries in them if I didn't mark them as kid game.

rbaird41, 2024-12-25:
Interesting. I'll keep tinkering with it and see if maybe I messed up
something somewhere. But since switching back to the full UI and restarting
emulationstation shows all the systems again, and since showing all games
while in kid mode shows all of the games. I'm pretty much at a loss. Thanks
for trying!

rbaird41, 2024-12-27:
Just an update on this. All of the systems show up on kiosk mode, so the issue only persists on kid mode. Not sure what that means, but wanted to add that.