christianhaitian/arkos #1239: Kodi Cache Settings not available / working in Kodi 21.1

allanguild, 2024-12-28:
Thank you for all the effort put into ArkOS - it's amazing, and runs brilliantly on an RG503 I picked up a couple of weeks ago.

I think I have found a small issue with cache settings in Kodi related to them moving from advancedsettings.xml to a GUI settings in the upgrade from 20 -> 21.

https://forum.kodi.tv/showthread.php?tid=377222

On the ArkOS version, I can't edit in advancedsetttings, the GUI page doesn't seem to be available and I can see in debug logging that the relevant settings parameters are not found and so default to 0 (which seems to be leading to some odd behavior when handling URLs) - Log excerpts below:

---

New Cache GUI Settings (replacement of cache in advancedsettings.xml) are:
                                                     Buffer Mode: 0
                                                     Memory Size: 0 MB
                                                     Read Factor: 0.00 x (adaptive)
                                                     Chunk Size : 0 bytes

----

debug <CSettingsManager>: requested setting (filecache.buffermode) was not found.

---

Is this something that could be addressed - I would also be happy to help (I have some coding and packagaing experience, but can't quite work out, how you are building Kodi).

Thanks again,

Allan

christianhaitian, 2024-12-28:
Oh interesting.  I released an OTA update a few days ago to revert Kodi back to 20.5 as i couldn’t get some addons to stream from urls.  This probably explains why.  The Kodi build recipe I use for ArkOS is here: https://github.com/christianhaitian/kodi-install/blob/master/ArkOS-Kodi-Build.sh

The Install directory is in /opt/kodi
The Kodi settings are in /home/ark/.kodi

christianhaitian, 2024-12-28:
Ah ha! When I updated Kodi to 21.1, I didn't update the settings.xml file located in the /opt/kodi/share/kodi/system/settings folder.  Once I added the missing Caching category there from here: https://github.com/xbmc/xbmc/pull/24049/files#diff-2ab50a0752d1209bc810232d63cbafdb5279e532632bf54c6d834951c4257602
It resolves the issue.  

Thank you again for your assistance in finding and solving this issue.  It was bugging me for weeks!

allanguild, 2024-12-28:
Great - thank you - updating that settings.xml file with that diff has fixed it for me as well.

Thanks again - really appreciate the quick response and all the work you do,

Allan

christianhaitian, 2024-12-28:
I'll have the fixed Kodi 21.1 install added in the next update.  In the meantime, if you feel the need, you can reinstall Kodi 21.1 with the necessary fixes by downloading the script in the link below and placing it in your tools folder then run it from Options>Tools in ArkOS.

https://github.com/christianhaitian/arkos/raw/refs/heads/main/scripts/Kodi-21.1-Install.sh

Thanks again.

closed 2024-12-28 by christianhaitian