christianhaitian/arkos #1240: RG353V - ArkOS_RG353V_v2.0_09292024 - After flashing image and inserting into device, RG353V boots directly to Android skipping ArkOS

bradleytenuta, 2024-12-30:
### Describe the Issue (If applicable)

1. Downloaded ArkOS_RG353V_v2.0_09292024 from Google Drive.
2. Formatted SD card of size 64gb from SanDisk to FAT32.
3. Used RUFUS to flash ArkOS_RG353V_v2.0_09292024 image to SD card.
4. Inserted SD card to TF1/INT on RG353V device.
5. Turning on RG353V results in Anbernic flash screen appearing then Android flash screen. Once done Android OS has loaded and able to use.

It appears that when the device is deciding what OS to boot, it is unable to discover the ArkOS installation on SD Card 1 and so boots Android OS instead.

### How can the issue be reproduced? (If applicable)
See Description.

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)
I am using device: RG353V.

### What version of ArkOS are you on?
ArkOS_RG353V_v2.0_09292024

### Anything else you'd like to include that may help to review this issue or feature request?
This issue appears regardless of what software I use to flash the OS.
- Rufus
- USB Image Tool

This issue appears even if I format the SD Card to FAT32 before flashing.

christianhaitian, 2024-12-30:
Are you able to boot the stock os without issue?  Did you make sure to not format the sd card after the image completed?

bradleytenuta, 2024-12-30:
Thanks for getting back! Yes, I'm able to insert the original SD Card and boot Stock Linux OS fine without issue. That's right I did not format SD Card after flashing the image.

christianhaitian, 2024-12-30:
You made sure to use 7zip to unpack the .img file from the .xz compressed file correct?  If so, try flashing the stock OS onto the Sandisk sd card using your same process and see if that works.  You can find the stock os here: https://win.anbernic.com/download/252.html

bradleytenuta, 2024-12-30:
I'll try flashing the stock OS onto the Sandisk sd card using your same process. Thanks again.

bradleytenuta, 2024-12-30:
So I was not able to successfully boot the stock OS on the Sandisk SD Card.

After flashing the Stock OS onto the SD Card, inserting it into the device and turning it on, I get the following sequence:

1. Anbernic flash screen for 1 sec
2. Then a blank screen for 1 sec
3. Anbernic flash screen for 1 sec
4. repeat

I've attached some photos to show how the SD Card looks after flashing it. Notice the unused space on the SD card is left unallocated.

I use 7zip to unpack the .ing.gz file: "right click -> "7-Zip" -> "extract to 'anbernicos-v12-20240120.img\'"

[image]
[image]
[image]

christianhaitian, 2024-12-30:
So something is wrong in your flashing environment.  Either you sd card writer is having issues or the sd card is faulty or something on your computer is causing an issue after the flash is completed.  Can you test an alternate sd card or card writer?

bradleytenuta, 2024-12-30:
Problem resolved when using a new SD Card.

closed 2024-12-30 by bradleytenuta