christianhaitian/arkos #1241: Love2d game engine issue with save game state

RobertoFreireFerrazPassos, 2024-12-31:
### Describe the Issue (If applicable)
filesystem save doesn't work for Love2d game

### How can the issue be reproduced? (If applicable)

I am creating a simple game and I am trying to save the score
It works fine in love-11.5-win64.exe
But when the save/load doesn't work in my handheld device

love.filesystem.newFile(scoreFile, "w") returns nil

```lua
local scoreFile = "highscoresnakegame.txt"

local function saveHighScore()
    local file = love.filesystem.newFile(scoreFile, "w")
    file:write(tostring(highScore))
    file:close()
end

local function loadHighScore()
    if love.filesystem.getInfo(scoreFile) then
        local file = love.filesystem.read(scoreFile)
        highScore = tonumber(file) or 0
    end
end
```

I also tried a different variation

```lua
love.filesystem.write(scoreFile, tostring(highScore))
```

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)

RG353PS

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

ArkOS_RG353M_v2.0_09292024.img

### Anything else you'd like to include that may help to review this issue or feature request?

RobertoFreireFerrazPassos, 2024-12-31:
It works with this code
```lua
local function saveHighScore()
    local file = io.open(scoreFile, "w")

    if file then
        file:write(tostring(highScore))
        file:close()
    end
end

local function loadHighScore()
    local file = io.open(scoreFile, "r")

    if file then
        local content = file:read("*all") 
        highScore = tonumber(content) or 0
        file:close()
    end
end
```

christianhaitian, 2024-12-31:
love2d is a mess with the current implementation in the OS.  I don't know how to solve it either since I don't understand how Love2D is supposed to work with Joysticks.  Also, Love2D engines don't seem to be backward compatible with some titles which is also confusing to me.

christianhaitian, 2025-01-05:
I've decided to deprecate support for Love2D since I can't get it to work for most games and I've lost interest in further working on it.

closed 2025-01-05 by christianhaitian

xanthiacoder, 2025-01-05:
Hi there! So sorry to add on to this after it's closed, but I do hope you'll reconsider. I'm actively developing for ArkOS and I hope to eventually release a full Digital Audio Workstation that runs on our ArkOS game consoles, and I'm doing it totally in Love2d.

I'm not sure about the struggles other developers are facing, but I can tell you that they are not insurmountable. You can see my working Drum Machine running on ArkOS here https://github.com/xanthiacoder/r36s-ledrums

If you could factor in my work to your considerations, please don't pull the plug on love2d. Or at least just leave it in as it is, since I can work with that.

If you need an active developer to work alongside with you for this matter, I am also open to it.

(btw I am releasing all my music app code under MIT, so you can also include that in ArkOS builds if you think it'll help with demonstrating that code can work on ArkOS)

xanthiacoder, 2025-01-05:
as a demonstration of everything working (save writes for persistence, all controllers mapped and functional)

8 tracks, 4 programmable loops, drum sequencer almost ready on #r36s #gameconsole #coding #love2d
https://youtube.com/shorts/vdJpAor0rX8?feature=share

christianhaitian, 2025-01-06:
I won't remove it from future installs.  I just won't actively support it.  If you can help figure out how to get more games to load and run with controls here, I might reconsider supporting it.  But for now, I can't figure out how to get games to load and run correctly.

xanthiacoder, 2025-01-06:
I'll do my best to help out, and respond to anyone that wants to figure out how to get their love code working.

For this explicit one regarding writing files, I know for a fact that the way to do it is to use LUA IO for writing. only use love2d FileSystem for reading. 

I am also watching love2d code and they are doing some things to their filesystem code. 

I've not had any issues with getting my code to work with R36S controllers. will watch to see if anyone needs assistance for controller issues.