christianhaitian/arkos #1242: Pico-8’s Splore deletes downloaded carts when played offline

elmimmo, 2025-01-01:
### Describe the Issue
With Wifi off, relaunching a Pico-8 game from Splore (v0.2.6b) from any section other than local storage overwrites its locally cached version with an empty file, rendering it unplayable.

### How can the issue be reproduced?

1. Make sure you start with no carts or BBS data stored locally. For that:
    1. Go to the Options category ▸ 351Files.
    2. Browse to `/roms2/pico-8`.
    3. Delete everything inside `carts` except `zzzsplore.p8` (if you don't have that file, create it so that you can launch Splore) and `bbs` (or else, move their contents elsewhere).
    4. Exit 351Files.
2. Turn Wifi on.
3. Launch Splore by going to EmulationStation’s Pico-8 category ▸ zzzsplore.
4. Inside Splore, switch to the “Featured” section and choose “[Update]”.
5. Launch any game listed there.
6. Exit Pico-8.
9. Turn Wifi off.
8. Verify that the game has been saved locally and is playable:
    1.  From within ES’ Pico-8 category, open the Options menu (Select button) ▸ Update Game Lists.
    2. The game should become listed inside ES’ Pico-8 category and be playable by choosing it (i.e. without entering Splore).
    3. Exit Pico-8.
9. Enter Splore again by going to ES’ Pico-8 category ▸ zzzsplore.
10. In Splore, switch to the local storage category (the icon of which is folder) and confirm the game is listed there, that its cover art is displayed when selected, and that the game is playable by launching it.
11. Return to Splore, switch to the “Featured” section where you originally played the game from, and launch the game. **An error will appear at the bottom saying “Could not connect to BBS.”**.
12. Exit Pico-8.
13. Verify that the game file has been overwritten with an empty file:
    1. From within ES’ Pico-8 category, launch the game (i.e. without entering Splore). **Pico-8 will boot and report that “Could not load /roms2/pico-8/carts/[game_filename]”**.
    2. Exit Pico-8.
    3. Go to the ES’ Options category ▸ 351Files.
    4. Browse to `/roms2/pico-8/carts`. **The game’s png file is now 0 kb**. That is confirmation no. 2.

### What device are you using?
RGB30

### What version of ArkOS are you on?
v2.0 (12/24/2024)

### Anything else you'd like to include that may help to review this issue or feature request?
* Using the 2nd SD for roms, hence the reference to `/roms2/…` instead of `/roms/…`. Did not try with the main SD for roms.
* [Pico-8 installed as decribed in ArkOS’ wiki](https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#pico-8).
* This issue does not happen with the same native version of Pico-8 on Rocknix (20241120) with the same device, [installed as described in Rocknix’ wiki](https://rocknix.org/systems/pico-8/). I.e. it does allow offline playing any game from any section (Favourites, Featured, etc.) that has been already played and therefore downloaded locally.
* I will try to report the same issue to Pico-8’s developer.

christianhaitian, 2025-01-01:
I see the issue but I'm not sure how I want to solve it yet.  To do it how ROCKNIX does it creates additional mess in the list of games in ES which I don't like.

elmimmo, 2025-01-01:
Do you have an understanding of what causes the issue, so as to assess whether it  makes sense that I do try to report it to Pico-8’s developer? Or is it really not something Pico-8 should or could address?

christianhaitian, 2025-01-01:
The issue is not with pico-8 but how it's launched.  I found a solution that seems to work and am doing some testing to confirm it should be good.

christianhaitian, 2025-01-01:
I believe I've found a resolve for this issue.  See if this works for you.

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1(TF1 Slot).  

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, trying launching splore and using it as you were before.

Let me know your results.

Thanks,

[arkosbackup.tar.gz](https://github.com/user-attachments/files/18286969/arkosbackup.tar.gz)

elmimmo, 2025-01-02:
It seems to fix the problem. Cached games in `/roms2/pico-8/carts/` now play fine from Splore's Featured and Favourites section with wifi off.

christianhaitian, 2025-01-02:
Great.  Thanks for testing and confirming the fix.

elmimmo, 2025-01-06:
Favourites now seem to be saved at `/opt/pico-8/favourites.txt` in the main SD card but not `/roms2/pico-8/favourites.txt` on a two card setup, as they where before the fix, which would be preferable for portability.

Same at least for:
* `/opt/pico-8/bbs/*` (except `carts` which seems to be symlinked or something)
* `/opt/*log.txt`

which I guess are less crucial but still.

christianhaitian, 2025-01-06:
I think favorites can be addressed for portability. Not sure about the rest or if it’s even worth worrying about.

christianhaitian, 2025-01-06:
Restore the file below following the same instructions from earlier [here](https://github.com/christianhaitian/arkos/issues/1242#issuecomment-2567183258)

This should symlink favourites.txt and sdl_controllers.txt to the roms card in use.

[arkosbackup.tar.gz](https://github.com/user-attachments/files/18319128/arkosbackup.tar.gz)

closed 2025-02-01 by christianhaitian