christianhaitian/arkos #1244: Aditional Drastic layouts

albertosf, 2025-01-02:
Is not possible to have aditional display layouts for Drastic emulator?

I have been using this machine and OS for a loong time and finished several GBA/SNES games. Now i am jumping to some NDS games and i am not very confortable with the screen layout+split.

Other OS they have some others

### Additional information:
Device: RG353M
Version of ArkOS 2.0 (09292024)
Drastic ArkOS version r2.5.2.2
Miyoo Drastic ArkOS version r2.5.0.4

[image]

christianhaitian, 2025-01-02:
There's a community mod that supports this but not sure if or when I'll include this.  You can find out more information on how to incorporate that into your setup here. https://github.com/Vidnez/Drastic-Patch-for-RGB30-20sx.  I don't know how well it will for the a 4:3 screen though so your mileage may vary.

albertosf, 2025-01-02:
Tried it, but not very sucesfull. 
Now not a single games is starting up and it goes back to UI.
It was a must to try ;)

closed 2025-01-11 by christianhaitian

agjino, 2025-09-04:
I managed to get this to work for my R36S. I replaced the factory version of Drastic with the one from this repo: [https://github.com/AeolusUX/ArkOS-R3XS](https://github.com/AeolusUX/ArkOS-R3XS). You'll have to access an ext4 filesystem, so it'll be much easier in a Linux environment. If your primary OS is Windows, you could put an Ubuntu installation in a virtual machine. Windows can access the EASYROMS partition in the sdcard, and you might make it work by placing the necessary files in EASYROMS first and using the device's File Manager to move them to the ROOT partition. But with Ubuntu, you'll have direct access to the ROOT partition from the PC.

Download an image of ArkOS from the repo above. I have the Feb 2025 firmware and Drastic from the July 2025 firmware did work. You'll have to unzip and mount the image. On Ubuntu, if you right click on a .img file and select "Open with other application..." it allows you to select Disk Image Mounter. Also insert the sdcard from the device. You'll obviously need an sdcard reader for this.

1. Open the ROOT partition both on the mounted ArkOS image and the sdcard
2. OPTIONAL: Write down user id and group id that have ownership of `/opt/drastic` and `/usr/bin/local/drastic.sh` on the sdcard
3. Rename the `/opt/drastic` folder on the sdcard to `/opt/drastic.bak`
4. Rename the `/usr/bin/local/drastic.sh` file on the sdcard to `/usr/bin/local/drastic.sh.bak`
5. Copy the `/opt/drastic` folder from the ArkOS image to `/opt` on the sdcard
6. Copy the `/usr/bin/local/drastic.sh` file from the ArkOS image to `/usr/bin/local` on the sdcard
7. IMPORTANT: Set +x (mark as executable) /usr/bin/local/drastic.sh and `/opt/drastic/drastic`
8. OPTIONAL: set the right ownership to /opt/drastic and `/usr/bin/local/drastic.sh`. You'll need to use the command line because the user/group don't exist on your machine so you'll need to use ids. On my device it was user:group 1002:1002
9. Copy `/EASYROMS/nds` from ArkOS image to sdcard. The subfolder `/EASYROMS/nds/bg` has 2 files. These will be used as layouts and you can switch between them. If you add more than 2 files Drastic will ignore the rest. You can only cycle between Single Scree, Layout 1 and Layout 2.