christianhaitian/arkos #1247: FreeJ2ME Could Not Start

tomasilavsky, 2025-01-07:
Hello,

I am using ArkOS on my RG351MP and attempted to try out the new update that added support for the J2ME emulator. However, I encountered the following issues with both cores:

- lr-freej2me: When I attempt to launch it, I get a black screen followed by an instant return to the menu.
- lr-freej2me-plus: When I try to use this core, I receive the following error message:
> FreeJ2ME Could Not Start, make sure frej2me-plus-lr.jar is in the system dir.

I followed both installation methods for JDK as described on the wiki:

1. Manual installation.
2. Automatic installation via the internet.

Unfortunately, both methods resulted in the same issues, and I was unable to get either core to work properly.

christianhaitian, 2025-01-07:
Do you have a `freej2me-lr.jar` and `freej2me-plus-lr.jar` in the bios folder of your roms or roms2 card?

christianhaitian, 2025-01-07:
Ok.  Someone else has reported this same issue via Discord.  Let me test this on my end later and see what the issue could be.  So far, it seems to be impacted rk3326 devices.

christianhaitian, 2025-01-08:
The issue was caused by a bad freej2me-lr.jar and freej2m2-plus-lr.jar file that was provided with the last update.  Restoring the attached backup file should resolve the issue for you now.  Please do the following and let me know if it does resolve the issue for you or not.

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1(TF1 Slot).  

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, trying launching your java games again.

Let me know your results.

Thanks

[arkosbackup.tar.gz](https://github.com/user-attachments/files/18340254/arkosbackup.tar.gz)

tomasilavsky, 2025-01-08:
Thank you for the instructions and the backup file. After restoring the settings, both cores are now working, and I can launch Java games without any issues.

However, I noticed that there is no audio in any of the games. Could you advise on how to resolve this?

Thank you again for your help!

christianhaitian, 2025-01-08:
Let me check something on this as it's only impacting rk3326 devices for audio.  I have an idea that I will test.

christianhaitian, 2025-01-08:
Hi,

Restore this file following the instructions from [here](https://github.com/christianhaitian/arkos/issues/1247#issuecomment-2576512461) and test audio for the java games.  This should include everything needed to fix java loading and sound for rk3326 devices.  Let me know your results.  

Thanks,

[arkosbackup.tar.gz](https://github.com/user-attachments/files/18352160/arkosbackup.tar.gz)

tomasilavsky, 2025-01-09:
Hi,

Unfortunately, after restoring the new file, neither core is launching java games anymore. This seems to be a different issue than before. Now, as soon as I try to launch a game, the screen briefly flashes and instantly returns to the menu.

christianhaitian, 2025-01-09:
> Hi,
> 
> Unfortunately, after restoring the new file, neither core is launching java games anymore. This seems to be a different issue than before. Now, as soon as I try to launch a game, the screen briefly flashes and instantly returns to the menu.

I just tested this and it works fine on my end.  Try restoring that file again and this time reboot your unit and see if that resolves the issue.

tomasilavsky, 2025-01-09:
I followed your instructions, restored the file again, and rebooted the unit. It seems the issue is now resolved. Thank you for your help!

christianhaitian, 2025-02-01:
This has been addressed with today's update.

closed 2025-02-01 by christianhaitian