christianhaitian/arkos #1250: bug for rk2023 （09292024）

lcdyk0517, 2025-01-11:
I found a bug in 2023, if I press the Reset button R on the es frontend interface, the machine will not be able to start up, and even if I press the shutdown button for a long time to make sure it is turned off and then turned on, it still doesn't work. I have to take out the battery and then insert it again to be able to turn it on. This kind of situation also occurs when switching to card 2 ROMs.
If I choose max in the overcharge dtb, the machine won't be able to turn on, and it won't work even if the battery is removed and reinserted.

christianhaitian, 2025-01-11:
1. The reset button should only be used if the system is unresponsive.  That can cause corruption if used regularly.
2. What sd card make and model are you using?  How did you apply the image to it?
3. If you're having frequent lockup and freezing, either you have a bad sd card or your unit just will not function well with ArkOS provided you're using a good sd card and properly imaged the sd card.

lcdyk0517, 2025-01-11:
I am using the official card of SanDisk that I use. To test whether this is a bug, I have rewritten the arkos09292024 package, then pressed the R key after booting into the system, and at this time the screen went black after the LED lights up. After waiting for a while, the problem persists. Therefore, I pressed and held the shutdown button to turn off the machine and then pressed it again to restart the machine, but the problem was still there. Then I reinserted the battery and could enter the system to start up normally.

I flashed the image using Rufus.

christianhaitian, 2025-01-11:
Why did you press the reset button after booting the device?  Was it stuck?

RetroGFX, 2025-01-11:
You're crashing your system on purpose, you should only use the reset if there's absolutely no other choice.. besides that shut down normally and you won't have any issues.. reset is for system locks etc

lcdyk0517, 2025-01-12:
I plugged in card 2, then used the sh to switch to card 2. After mounting roms2, arkos would reboot, but at this time, arkos also could not start up properly, only displaying green lights. This situation is exactly the same as when I press the R key. Of course, there is no problem with my card 2; I used rgb20pro's arkos to successfully switch to card 2.

christianhaitian, 2025-01-12:
After a fresh flash, did you update to the latest ota update which is 12242024?

lcdyk0517, 2025-01-13:
Currently not, I'll try later

lcdyk0517, 2025-01-20:
Recently, I found in the rk2023 chat group that many friends have encountered the same situation as me, but a few friends have no problems using arkos. It should be that this machine has multiple versions with different power management chips. Unfortunately, the arkos on my machine has bugs.

christianhaitian, 2025-01-27:
Closing this issue as not much I can help with since I can't replicate the issue on my end.  It could be that ArkOS is just not compatible with the specific unit which is non uncommon for some rk3566 devices unfortunately.

closed 2025-01-27 by christianhaitian