christianhaitian/arkos #1252: Small Update of /usb/bin/msgbox

XNLFutureTechnologies, 2025-01-12:
Hey, first of; Great job on this OS!

I have a small request, and have already been looking through the files of both the ArkOs repository and the Linux kernel repository in your account (because otherwise I could have opened a pull request), but I could not find the file in question there (sorry).

It's for the file: /usr/bin/msgbox:

I would like to request/suggest an small update on line 49 by changing it from:
`dialog --clear --msgbox "$1" $height $width 2>&1 > /dev/tty1`

to:
`# If a second argument (a dialog title) is given, well add it as output
if [ -n "$2" ]; then
  title="$2"
  dialog --clear --title "$title" --msgbox "$1" $height $width 2>&1 > /dev/tty1
else
  # If no dialog title is given we'll just use the orignial line
  dialog --clear --msgbox "$1" $height $width 2>&1 > /dev/tty1
fi`

I'm currently using both the R36S and the R36H both running ArkOS 2.0 12242024

I know it might be a small thing or even considered useless to some, but I'm currently working on add-on scripts which do use the title in an 'advanced' dialog system 'application script' which will make it easier for non linux users to install additional controllers, options and features etc if they want or need to. And the dialog title can/will make it significantly easier to have good indication about in what script gave the message. While my change will still be 100% compatible with scripts that don't use the title :)

If rather have me make a pull request of it, just let me know where I can find the particular script to change/add it and request it :)

christianhaitian, 2025-01-14:
Hi

The latest version of that script is [here](`https://github.com/christianhaitian/rk3566_core_builds/blob/master/shell-scripts/msgbox).  I meant to include a copy of the script in the rk3326 repo as well.  You can submit your pull request for it there.  Also note, I don't support the r36s unit but I believe the community maintained image includes updates I push out typically monthly so it should eventually make it's way there and some point.

XNLFutureTechnologies, 2025-01-14:
Thanks! Will make a pull request for it there :)

I know I have seen the community version, but personally prefer yours to be honest 😊
It works perfectly fine, and with some add-on scripts & plugins I'm currently working on it will be 'top notch' for both the R36S and R36H.  Currently working on some system tools, package installers and additional hardware support which people can just install with simple sh scripts dropped into the tools folder. And everything will be user friendly using dialogs for them.

Hence why I noticed the missing title part 😉 

Your version runs just super smooth. And for some reason I had some issues with my R36H (not the S) using the community image, while yours just runs super on both. So thanks for your reply and your awesome job on this project.