christianhaitian/arkos #1257: Hello, I need help again. Vibration Issues with RK2023

retrooldman, 2025-01-17:
Hello, I need help again. I recently bought the black WiFi version of the RK2023 and installed the latest ArkOS. Just like the RGB30, its motherboard has a vibration module. I soldered a vibration motor and configured it according to the RGB30 settings, but neither GBA games nor games like Tekken 3 on PS support vibration. In the Frequently Asked Questions for RK2023, there's mention of game vibration, but I'm not sure if the RK2023 hardware supports it.         ps.This translation comes from Musk's AI GROK .‘）

[image]
[image]

christianhaitian, 2025-01-17:
When you enabled the vibration Setting and exit the menu, did the motor vibrate?  It should have.  If it didn’t, try the controller tester in the Options>Advanced section and press r3 and l3 together and see if the motor vibrates. If it doesn’t, recheck the motor itself and the soldering job.  If the motor is working fine in the OS per my suggested testing above, update the emulators via retroarch and try your games with vibration again

retrooldman, 2025-01-17:
There is no 'Enable the Vibration Setting' in the Options>Advanced menu of the RK2023

[image]

retrooldman, 2025-01-17:
Wow, amazing! Following your instructions, the vibration motor soldered onto the RK2023 is working. Thank you so much! 

[image]

P.S. It needs a restart.

christianhaitian, 2025-01-17:
Thanks for confirming.  I'll include this in the next update as well.  Enjoy!

retrooldman, 2025-01-17:
Thank you once again ;)

christianhaitian, 2025-02-01:
This has been added with today's update.

closed 2025-02-01 by christianhaitian

retrooldman, 2025-02-04:
https://github.com/user-attachments/assets/8a36921b-bf9b-46e4-8378-93dfd307444b

For the RGB30 and RK2023, after installing the latest system update from 02022025, the vibration module is enabled in the advanced settings, but there is no vibration feedback when exiting. On the RK2023, RetroArch32 crashes and cannot be opened, and PS games cannot be loaded

christianhaitian, 2025-02-04:
Looks like with the updated kernel that now supports native rumble, enabling and disabling the optional rumble motor is no longer necessary.  You can still test vibration using the Controller Tester available in the Options>Advance section.  Just press L3 and R3 with that app and the motor should still vibration if functioning correctly.  I can't replicate the retroarch32 not launching issue on my end.  After the latest update, it launches just fine on my end.  Perhaps a new image install and update is needed.

retrooldman, 2025-02-05:
After reinstalling the 02012025-3 version without performing an OTA update, retroarch32 now works properly. Could it be that the OTA update to version 0202 caused the issue with retroarch32?

retrooldman, 2025-02-05:
The vibration works normally during gameplay.

retrooldman, 2025-02-05:
The latest 0202 version of rgb30 ota can be used normally with retroarch32.