christianhaitian/arkos #1262: Does boxart scraping only work for non-zipped ROMs?

mainrs, 2025-01-27:
I bought a ROM on itch.io that is also present on thegamesdb: https://thegamesdb.net/game.php?id=130225

However, ArkOS doesn't find it. Browsing the game list and selecting Powa! within the GBC menu does not show the boxart. My ROM is zipped. Is that the issue?

Thanks!

christianhaitian, 2025-01-27:
I'm not 100% sure but you could try it unzip and see if you get a different result.  How were you trying to scrape for it?

mainrs, 2025-01-28:
Using Start button -> Scrape -> Scrape now. 

I also unzipped the ROM and selected thegamesdb for Powa!, because the other one has no boxart for it. Still, nothing was found.

christianhaitian, 2025-01-31:
I finally got a chance to test this.  I pressed Start button -> Scraper.  Then selected the TheGamesDB, then Scrape Now.  The game scraped just fine and loaded the Box art on my end.  The file name is `Powa! (World) (En) (GB Compatible) (Aftermarket) (Unl).zip`.

[image]

closed 2025-02-01 by christianhaitian