christianhaitian/arkos #1266: portmaster won't launch after updating to v2.0 (02/01/2025-2)

lourinaldi, 2025-02-01:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANce FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

Yes

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

Yes

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

Yes

### Describe the Issue (If applicable)

portmaster won't launch after updating to v2.0 (02/01/2025-2). the update also appears to have broken at least some installed ports.

### How can the issue be reproduced? (If applicable)

After updating to v2.0 (02/01/2025-2), attempt to launch portmaster.

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)

RGB30

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

v2.0 (02/01/2025-2)

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2025-02-01:
Not sure why the update would have caused that issue.  I can't figure out why it's segfaulting.  I've reported it to the PortMaster team so hopefully then can help figure out the cause.

christianhaitian, 2025-02-01:
The issue has been identified and a fix will be released shortly.

christianhaitian, 2025-02-01:
This has been resolved with the update released just now.  Thanks for reporting the issue.

closed 2025-02-01 by christianhaitian