christianhaitian/arkos #1267: State Slot could not be set to Auto and Error removing overrides

Piggy8013, 2025-02-03:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANce FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?
Yes

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?
Yes

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?
Yes

### Describe the Issue (If applicable)
I tested GB,GBC and GBA games, and the State Slot could not be set to Auto. Every time I set it to Auto, save the independent overrides, exit the game, and open it again, the status card slot will return to 0.
Another problem is that the Remove Core Overrides prompt "Error removing overrides".

### How can the issue be reproduced? (If applicable)
State Slot could not be set to Auto, Every time I set it to Auto, save the independent overrides, exit the game, and open it again, the status card slot will return to 0.
Remove Core Overrides prompt "Error removing overrides".

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)
RGB30,RG353VS,RG353PS

### What version of ArkOS are you on?
v2.0 (02/02/2025)

### Anything else you'd like to include that may help to review this issue or feature request?

I like ArkOS very much. It is very easy to use and brings many surprises. I think it is the best system. 

I have always kept the system up to date, but after the v2.0 (01/31/2025) update, I found a small problem. I tested GB,GBC and GBA games, and the State Slot could not be set to Auto, Every time I set it to Auto, save the independent overrides, exit the game, and open it again, the status card slot will return to 0.

Another problem is that the Remove Core Overrides prompt "Error removing overrides".

Never had these problems before.

Thank you again for everything you do.

christianhaitian, 2025-02-03:
I was able to recreate the issue and I suspect it's an issue libretro will need to resolve.  I am able to save core overrides for nestopia for instance.  When I exit Retroarch then open a nestopia compatible rom via retroarch, it opens, it says it appends the core override, but the settings are not there that should be there.  They are in the override config but are not applied.  If I manually load them, it applies the override fine.  It's just not truly doing this upon launch of the core even though it claims that it does.  

I've opened a ticket with libretro on the issue here: https://github.com/libretro/RetroArch/issues/17510

christianhaitian, 2025-02-03:
I can't replicate the issue with not being able to remove core overrides from the menu though.  That works just fine on my end.  If you regularly access the OS via ssh and make changes, try making sure the /home/ark/.config/retroarch folder is owned by the ark user.  You can do this from ssh by doing the following: `sudo chown -R ark:ark /home/ark/.config/retroarch*`

Piggy8013, 2025-02-03:
> I can't replicate the issue with not being able to remove core overrides from the menu though. That works just fine on my end. If you regularly access the OS via ssh and make changes, try making [...]

Thank you for your reply. 
I followed your instructions, but the error persists. This time I tried to delete the mGBA directory under /home/ark/.config/retroarch/config, but failed. I suspect it is a permission issue. I tried to delete and overwrite the previous version without any problem. (System before v2.0 (12/24/2024))

christianhaitian, 2025-02-07:
The override issue does not occur on the stable retroarch release of v1.20.0.  I will provide an update to this version as soon as possible.

Piggy8013, 2025-02-08:
> The override issue does not occur on the stable retroarch release of v1.20.0. I will provide an update to this version as soon as possible.覆盖问题不会出现在 v1.20.0 的稳定 retroarch 版本上。我将尽快提供此版本的更新。

Great! I'm glad to hear this news. This will be the happiest thing for me recently.

christianhaitian, 2025-02-08:
This has been addressed with today's update.

Piggy8013, 2025-02-09:
> This has been addressed with today's update.

Thanks again for everything!
But I updated two of my devices(RGB30 and RG353PS) to the latest version v2.0 (02/08/2025),State Slot still could not be set to Auto,I've tried countless times, including restart, and I'm about to crash!
Sorry to bother you again.

christianhaitian, 2025-02-09:
Oh my, you're absolutely right.  I somehow didn't build the right version.  I'll get this corrected shortly.

christianhaitian, 2025-02-09:
This should be taken care of properly with today's update.  Thanks again for reporting the issue.

closed 2025-02-09 by christianhaitian